laboshinl/loam_velodyne #3: Public place to ask questions, post and discuss issues

laboshinl, 2016-04-11:


laboshinl, 2016-04-11:
The mapping is working, but in some point the pointcloud suddenly returns to the initial position and I get this error:

  ```
  laserOdometry: /build/buildd/pcl-1.7-1.7.1/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:136: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const [with PointT = pcl::PointXYZI; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.
    [laserOdometry-2] process has died [pid 10107, exit code -6, cmd /home/vicky/catkin_ws/devel/lib/loam_velodyne/laserOdometry __name:=laserOdometry __log:=/home/vicky/.ros/log/580f7db4-ffc0-11e5-aee2-74d4358625bb/laserOdometry-2.log].
    log file: /home/vicky/.ros/log/580f7db4-ffc0-11e5-aee2-74d4358625bb/laserOdometry-2*.log
    [laserOdometry-2] restarting process
```

I think when a keypoint from the registration process is not found, then the node laserOdometry node crashed and restart, then the mapping does not work because it does not continue in the same position as before, however it start again from the initial position.

When you run your example:
```
roslaunch ~/catkin_ws/src/loam_velodyne/launch/loam_velodyne.launch
rosbag play ~/Downloads/nsh_indoor_outdoor.bag 
```
do you obtain the same error?

How can I avoid this error? Many thanks in advance.

laboshinl, 2016-04-11:
> Is this a version you wrote or is it the original implementation or did you piece it back together from the help pages?
> Did you write the package and launch file?

I've created package and launch file. I've also modified scanRegistration.cpp to get it to work with my VLP16 (I think originally it was for HDL32 )

ferrefrederick, 2016-04-26:
Do you know if it saves the map generated anywhere? and if not, are you planning on modifying it so it does?

DamonMIN, 2016-04-27:
When I build with catkin "source ~catkin_ws/devel/setup.bash", there is an error "bash: ~catkin_ws/devel/setup.bash: No such file or directory". Please help me to solve it. Thank you. @laboshinl

ZepherusFF, 2016-04-27:
Hi @DamonMIN did you setup your catkin workspace correctly based on the ROS documentation ? Do you have the file in that directory ?

DamonMIN, 2016-04-27:
I have the "setup.bash" file in the folder ~catkin_ws/devel. The following text is the content of this file.
////////////////////////////////////////////////////////////////////
#!/usr/bin/env bash
# generated from catkin/cmake/templates/setup.bash.in

CATKIN_SHELL=bash

# source setup.sh from same directory as this file
_CATKIN_SETUP_DIR=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)
. "$_CATKIN_SETUP_DIR/setup.sh"
///////////////////////////////////////////////////////////////////

laboshinl, 2016-04-27:
@DamonMIN  You have missed "/"
```
laboshinl@rtc ~/catkin_ws $ source ~catkin_ws/devel/setup.bash 
bash: ~catkin_ws/devel/setup.bash: No such file or directory
```
```
laboshinl@rtc ~/catkin_ws $ source ~/catkin_ws/devel/setup.bash 
```

DamonMIN, 2016-04-27:
@laboshinl 
###In second terminal play sample velodyne data from VLP16 rosbag:
rosbag play ~/Downloads/2014-11-10-10-48-31_Velodyne-VLP_20Hz_Hecker Pass.pcap
[ INFO] [1461760124.789037514]: Opening /home/aicrobo/Downloads/2014-11-10-10-48-31_Velodyne-VLP_20Hz_Hecker
[FATAL] [1461760124.789397586]: Error opening file: /home/aicrobo/Downloads/2014-11-10-10-48-31_Velodyne-VLP_20Hz_Hecker

### read from velodyne VLP16 sample pcap:
roslaunch velodyne_pointcloud VLP16_points.launch pcap:="/home/aicrobo/Downloads/2014-11-10-10-48-31_Velodyne-VLP_20Hz_Hecker Pass.pcap"
[VLP16_points.launch] is neither a launch file in package [velodyne_pointcloud] nor is [velodyne_pointcloud] a launch file name
The traceback for the exception was written to the log file

It seems that the launch file can't find or recognize the velodyne data(The data was downloaded from VLP16 sample pcap).

laboshinl, 2016-04-27:
You should compile velodyne driver
https://github.com/ros-drivers/velodyne/tree/master/velodyne_pointcloud

ZepherusFF, 2016-04-27:
@DamonMIN  For the VLP16 you need to compile yourself the driver

akshayshetty, 2016-05-27:
Hi,

Can you mention what changes you made to the original code? Did you just change the scanRegistration.cpp?

Also did you add/remove any header files?

Thank you!

Fazt, 2016-05-30:
Is there any way to export the data or is just for visualization?. By the way, works great.

laboshinl, 2016-06-02:
@Fazt 
You can save registered clouds to pcd and open with ccViewer
**rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered**

drscopus, 2016-06-03:
It seems that I can't run it successfully.  I used the "VLP16 sample pcap" data sets. Should I calibrate the Velodyne sensor at first? 
 At first, the 3D data points couldn't be displayed in rviz, only two odom topics /laser_odom_to_init and /integrated_to_init could be displayed. The locations of these odom topics are strange. Initially, they start at a same position. They are getting further in the distance. And the orientations of these two topics are also very strange. 
 Then I found if the displayed style of pointcloud2 is adjusted from "points" to "squares", the 3D data points could be displayed. However, I couldn't see the constructed 3D map in rviz.
Looking forwards to any great advices. 
Best!

laboshinl, 2016-06-03:
@drscopus 
Did you clone latest version of repo? It has different rviz config.
You should look only at RED odom lines. Just disable GREEN. Arrows are perpendicular to the the direction of movement for some reason.

laboshinl, 2016-06-03:
@drscopus 
> I couldn't see the constructed 3D map in rviz.

Try to set '**Decay Time**' parameter to **1000** in topic '**velodyne_cloud_registered**'

drscopus, 2016-06-04:
@laboshinl Thanks! It seems works fine now!

CansenJIANG, 2016-06-15:
Dear Leonid,

Hello, I'm a Phd student from Univ. de Bourgogne in France. And recently I found your awesome project on github using Zhang's method on SLAM. I tested your code and it was working very well. However, I have some questions on the performances of the registration results. I tested the dataset that you shared on Github (VLP16 rosbag). Here are the results that I obtain (Fig.1 is the top view, and Fig.2 is the side view) . As we know, the Zhang's method achieves the best performances among the KITTI dataset (the LOAM ranked second, while V-LOAM ranked first). But the registered clouds that I got from the dataset that you provided are bent (see Fig.2). 

I would like to ask that if you are having the same problem on this testing data or not? Or do you know the reason why it happens?

Thank you very much and look forward to your reply!
Cansen

[image]
Inline image 1
[image]
Inline image 2

haoala, 2016-06-16:
Hi,

The LOAM paper defines a formula for computing curvature that normalises by the distance of the point from the origin (equation (1) in the paper). However, it seems like the curvature computed in line 378 of scanRegistration.cpp is not normalised by the distance of the point from the origin. Any idea why this might be the case?

Thanks!

TopGunSnake, 2016-06-17:
@laboshinl When testing the loam_velodyne with real data, I noticed that the transform frames are in a non-standard coordinate system. Namely, sensor forward is mapped to +z, sensor up is mapped to +y, and sensor left is mapped to +x. Is there a reason for this, and where can I easily change it to standard cartesian?

haoala, 2016-06-18:
@TopGunSnake I believe it's because that's the convention the paper uses (page 2 at http://www.roboticsproceedings.org/rss10/p07.pdf).

haoala, 2016-06-24:
@laboshinl What units are distances in?

Kailegh, 2016-06-30:
For the error generated in nsh_indoor_outdoor.bag  is it possible to make that when it losses odometry it keeps the last pose?
And I have other question, is it possible to add the help of an IMU or wheel odometry?

laboshinl, 2016-06-30:
@Kailegh There is no error for me with current version

Kailegh, 2016-06-30:
> @Kailegh There is no error for me with current version
60db8c1

Thanks a lot, works perfect now I had an old version sorry. Great work man!

Regarding the IMU, do you think there is any advantage of including one?

laboshinl, 2016-07-01:
@Kailegh I have not tested it with IMU yet. 
https://github.com/laboshinl/loam_velodyne/issues/6

haoala, 2016-07-04:
I believe there's a bug with laserOdometry.cpp. Lines 458-459 should really go after line 470. You want to reset the points used to build the Jacobian at the start of each new iteration, and not keep points from previous iterations.

laboshinl, 2016-07-12:
@haoala 
Feel free to send pull requests with updates

doublej317, 2016-07-15:
Can you share the original version of scanRegistration.cpp that works with HDL-32E? Thanks.

laboshinl, 2016-07-15:
@doublej317 I was mistaken, it doesn't seems to be for HDL64. Original sources can be found here: http://docs.ros.org/indigo/api/loam_velodyne/html/files.html

doublej317, 2016-07-15:
@laboshinl  Thank you :)

haythamd, 2016-07-27:
Hi thank you for this project LOAM seems amazing for mapping,

 I am new to linux but I followed your steps but when I try to launch loam_Velodyne using "roslaunch ~/catkin_ws/src/loam_velodyne/launch/loam_velodyne.launch,"
 it gives me the following error
  " raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: loam_velodyne"
I build the catkin and all the files are there, I also built velodyne vlp 16 drivers ,
any advice on how to solve this? thank you

laboshinl, 2016-07-27:
@haythamd I'm not really sure, but it looks like a problem with environment variable, have you executed the following command? 
`source ~/catkin_ws/devel/setup.bash`
Is your ROS configured correctly?

haythamd, 2016-07-27:
@laboshinl  thank you for you fast reply, I did the steps all over again and it now worked but only If I used the command in the tutorial in cd ~/catkin_ws, anyway it works now and opens RVIZ but when I opened a new terminal and used the command "roslaunch velodyne_pointcloud VLP16_points.launch pcap:="/home/laboshinl/Downloads/velodyne.pcap" with changing the directory name and file ofcourse to play a sample VLP 16 pcap file that I downloaded from Velodyne it gives the following error: 
"[VLP16_points.launch] is neither a launch file in package [velodyne_pointcloud] nor is [velodyne_pointcloud] a launch file name
The traceback for the exception was written to the log file"
I already installed velodyne 16 drivers from source is there another driver I should install? 

Thank you

laboshinl, 2016-07-27:
@haythamd Oh, I think you should also execute
`source ~/catkin_ws/devel/setup.bash` in second terminal window to make you self-compilled velodyne driver to work

haythamd, 2016-07-27:
@laboshinl yes I think you were right , it now lets me run the command however it runs into several errors then stops, I get the following errors: 
" Error opening Velodyne socket dump file.
failed...
failed...
 all processes on machine have died, roslaunch will exit
shutting down processing monitor..."
Do you think it would be better to try convert it to a bag file and try the other command? or could the sample data from velodyne be corrupt? 

Thank you

haythamd, 2016-07-27:
@laboshinl I was able to run the pcap file by changing the command a bit but I dont see a 3d map all I see is red arrows and I even tried increasing the decay Time,  I took a screen shot of it below 
[image]

Thank you for all your help , do you have any suggestions to improve it?

haythamd, 2016-08-05:
@laboshinl  I used the command" rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered"
and got alot of .pcd files , I then downloaded cloudcompare but it says it cant open pcd files, any suggestions? do I have to install a certain plugin for it to work? thank you

CansenJIANG, 2016-08-06:
@haythamd 
I suggest you read the point cloud using PCL or Meshlab, it's much more convenient. However, if you need to using some functions of cloudcompare, feel free to go. Just check [the report](https://ceen.et.byu.edu/sites/default/files/snrprojects/wolfe_derek.pdf) and follow step by step. 

Cansen

ctwu0314, 2016-08-08:
@laboshinl
I have successfully implemented according to your steps.
But when I connect the VLP-16 Lidar (roslaunch velodyne_pointcloud VLP16_points.launch _ip_address:=192.168.1.201),
Usually happen the following error,

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
[scanRegistration-2] process has died [pid 13615, exit code -11, cmd /home/ctwu/ROS_Code/catkin_loam/devel/lib/loam_velodyne/scanRegistration __name:=scanRegistration __log:=/home/ctwu/.ros/log/1f7faff2-5d33-11e6-98e2-74d02b61e376/scanRegistration-2.log].
log file: /home/ctwu/.ros/log/1f7faff2-5d33-11e6-98e2-74d02b61e376/scanRegistration-2.log
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
(the scanRegistration-2.log can't find in the folder)

arjunmenon03, 2016-08-18:
@TopGunSnake @haoala I noticed the non-standard coordinate frames and thought maybe it was LOAM that was changing the coordinate frames but when you stream Point cloud data from the LiDAR and display it on Rviz, it still has the same coordinate frames as we see on LOAM. Were you guys able to fix the coordinate frames by writing a tf?

TopGunSnake, 2016-08-18:
I had to ditch this node due to time constraints, but once I finish my
project, I will be looking at this node again.

Jake Webster

haoala, 2016-08-19:
For my application, the lidar data was coming in using the coordinate
system I wanted, so I didn't have to change it there. In any case, using a
different coordinate frame doesn't change the algorithm, and it shouldn't
be too hard to change the coordinate frame of the incoming lidar data.

haythamd, 2016-08-19:
I have been testing out LOAM mapping for some time now with different applications with my VLP-16, for some reason RVIZ doesnt show any points or clouds until I change the current view to topdown orthogonal , does anyone know why this happens? also my mappings don't really come out as clean and crisp as the sample pictures, any advice on how to improve them?

desmosedic, 2016-08-19:
@TopGunSnake @haoala I wrote a simple fixed frame that ended up putting it in the same coordinate frame as the Velodyne. Quaternion of [0, 0.7071, 0.7071, 0] will put it in the Velodyne's coordinate frame. You are right @haoala , this does not change the algorithm whatsoever.

wine3603, 2016-08-25:
Hi guys, I am using LOAM too! 
are there anybody use LOAM on Hokuyo ? I had some problems on synchronization of laser and imu.

TopGunSnake, 2016-08-28:
@desmosedic What frames was the transform for?

cuipengfeily, 2016-09-08:
@laboshinl ,@haoala ,Hi, I have run the command "rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered" and then I open them in CloudCompare,but it seems to be not registrated.However, the topic "laser_cloud_surround" is the all registrated points. So would you like to help me how the topic "laser_cloud_surround" can be saved in pcd format.Thank you!

ShineSong, 2016-10-12:
@Fazt 
I recommend this package to export point cloud data. https://github.com/ShineSong/gatherpclmsg

Kailegh, 2016-10-18:
I know this is asking a lot, but could you write some comments in the code? I have tried to understand how it works, but is everything too messy and I do not really get it. 
Thanks a lot anyway for your code

amenonDJI, 2016-10-19:
@cuipengfeily It would be the same process to save the laser_cloud_surround into a bag file and then use bag to pcd. I had no issues doing that. 

@laboshinl I had a related question for you - when I set decay time to 1000, I get a beautiful map on rviz. But this is only for visualization purposes. What if I wanted to convert this to PCD and view it in CloudCompare tool, how can I make the point cloud persistent? If I take the bag file and convert it to PCD now, the map looks similar to the visualization in rviz with decay time of 0.

laboshinl, 2016-10-20:
@amenonDJI 
` rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered` saves .pcd for each registered pointcloud. Then this files should be simultaneously opened with CloudCompare

cuipengfeily, 2016-10-20:
@amenonDJI I have tried with your help,and I have no issues now.
Thank you

desmosedic, 2016-10-20:
@laboshinl That makes sense. I opened them simultaneously and they work fantastic. In fact I took some measurements to see what kind of error I was receiving and I was able to measure with less than <2% of error with measurements. Do you have any error % from your tests. LOAM paper lists a few measurements as well.

DamonMIN, 2016-10-22:
@laboshinl When I run "roslaunch loam_velodyne loam_velodyne.launch", there is an error as follow shows:
[image]

desmosedic, 2016-10-22:
@DamonMIN As a debugging step, remove the call to Rviz inside the launch. Run the LOAM node separately and call rviz in another terminal. Then choose the laser_cloud_surround topic.

desmosedic, 2016-10-22:
@laboshinl @TopGunSnake @CansenJIANG   I have some test accuracy data and the algorithm is looking really good. I have less than 2% error in all the geometric calculations. There is a small problem though, a lot of the dynamic obstacles in the map such as a human walking around show up as artifacts in the map. Is there a way to reduce the "sensitivity" per say, in order to filter out point cloud data that are not static? Are there any parameters we can tune?

JohnMMeyer, 2016-10-23:
Hello,
 I have just discovered LOAM, it seems very useful. I had a question if you don't mind. I have VLP-16 data that I collected a few months ago. They are in .bagfiles with the following topics:

topics:      /cloud_in                               1969 msgs    : sensor_msgs/PointCloud2              
             /cloud_nodelet/parameter_descriptions      1 msg     : dynamic_reconfigure/ConfigDescription
             /cloud_nodelet/parameter_updates           1 msg     : dynamic_reconfigure/Config           
             /diagnostics                             197 msgs    : diagnostic_msgs/DiagnosticArray      
             /rosout                                   19 msgs    : rosgraph_msgs/Log                     (5 connections)
             /rosout_agg                                2 msgs    : rosgraph_msgs/Log                    
             /tf                                     2062 msgs    : tf2_msgs/TFMessage                   
             /velodyne_nodelet_manager/bond           828 msgs    : bond/Status   (3 connections)              
             /velodyne_packets                       1969 msgs    : velodyne_msgs/VelodyneScan

Is it possible to build a map with these topics? In other words without specifically having the topic 'velodyne_cloud_registered' ? 

I have pulled out the pcd files already, can I build a map using those?

Thank you very much

laboshinl, 2016-10-23:
@JohnMMeyer 
I think you should remap your **cloud_in** topic  to **velodyne_points**
http://wiki.ros.org/roslaunch/XML/remap

JohnMMeyer, 2016-10-25:
@laboshinl , thank you for the input. I did convert cloud_in to velodyne_points and that is playing nicely in RVIZ. My ultimate goal is to create a map and display it in Cloud Compare. I have been looking around and can't seem to find a way to do that with just the bagfile I have. Any help would be great. 

@Durant35 I looked at the link you sent, thanks for that. Which paper specifically were you referencing/

Thanks for the help,
John

[129 more replies]