laboshinl/loam_velodyne #6: Adding IMU doesn't seem to work

droter, 2016-06-21:
When I try to add an IMU topic it doesn't seem to change the output.

Anyone get this working with an IMU?

Thanks,

Matt

tiralonghipol, 2017-11-10:
Hey! Have you tried to check the right name of the Imu topic ?
I had the same issue and the problem was simply that my simulation was publishing on "Imu/" 
but the algorithm want in input "Imu/data/"

rcabg, 2018-02-19:
Hey, same problem with IMU here.

Using just pointcloud data seems to work well. But trying to add an Imu is freaking me out. Output becomes completely crazy. Imu is aligned with my Velodyne and both use ROS axis references (x-front, y-left, z-up). Any ideas of what could be wrong? 

Thanks in advance!

2amour, 2018-03-31:
Hey!The same problem with IMU.
Using the code stated in https://github.com/laboshinl/loam_velodyne/ run the gates_oscillating_motion.bag can not get the results like the demo in ROS wiki 
Anything wrong?

laizwithzed, 2018-08-09:
Hello everyone,
I have the same problem as you, @rcabg . I changed the axis references according to the settings of my mounting in ScanRegistration.cpp and published the transformed IMU messages as /imu/data to be used in LaserMapping.cpp. Everything seems fine with this adaptation, but the algorithm simply goes crazy.
Any ideas of what could be wrong?
Thanks in advance!

rcabg, 2018-08-17:
Hey @laizwithzed

Check imu data consistency (for instance, check that timestamps are correct. Also I had sometimes two imu messages with the same header. Driver node problems...)

If imu axis are aligned correctly with the lidar, try to record a Ros bag with lidar+imu and then play it back slower (use "--clock" param as well as "-r 0.5" to make it slower).

I realized that my computer wasn't powerful enough to process everything in real-time but it performed correctly when I was using a rosbag with slower speed.

Tell me if you found this helpful,
Cheers

laizwithzed, 2018-08-20:
Hi @rcabg,

Thanks for your comments, but unfortunately it didn't work.

I had to transform the orientation axis in my code to allign the IMU data and Velodyne points. To do so, I applied a rotation+translation transform in the coordinates from the topic /an_device/Imu/ (in ScanRegistration.cpp) and published the transformed coordinates in the topic /imu/data/ (that I'm using in LaserMapping.cpp).

I tried your suggestion to play the bag file at a slower speed, but it didn't work, as the algorithm goes crazy in slow motion.

Did you change something in your code?

rcabg, 2018-08-20:
Hi @laizwithzed

No, just lidar+imu alignment. Also, does your linear acc in axis Z include gravity? It has to. It could be weird, but I've seen some IMUs that don't.

laizwithzed, 2018-09-04:
Hi @rcabg,
In fact, my IMU data didn't contain any acceleration at first and it seems
to be the cause of the problem, as the algorithm worked fine with new bag
files including acceleration data.
Best,
Laiz

BrunoEduardoCSantos, 2018-09-26:
Hi @rcabg , 
could you detailed how did you align both IMU and LiDAR?
Thank you.
Best,
Bruno

w4rlock999, 2018-10-03:
@laizwithzed how do you do the IMU? what are the axes convention you use?

luhcforgh, 2018-10-05:
Hi guys! I just wanted to check in to keep the conversation going, since we seem to be having similar issues.

My team is working on creating an autonomous rover using a Velodyne Puck VLP-16, Zed camera, and a BNO080 IMU.  We're evaluating LOAM for now, to see if we can perhaps fuse the output with a camera based SLAM algorithm later (ORB-SLAM2 is showing promising results). As we're new to ROS and SLAM, please point out any rookie mistakes that we might be making. 

Unfortunately, we are also struggling with getting LOAM working together with the IMU, even at standstill. I'll make sure to provide a detailed description of what we're seeing the next time we have it all set up. So far, we have tried:

- Adding/removing gravity from the accelerometer data (including gravity works OK, but not better than pure LIDAR). The gravity is actively removed after being received in https://github.com/laboshinl/loam_velodyne/blob/master/src/lib/ScanRegistration.cpp, line 171-3.
- Adjusting the name of the expected Imu message to /imu_raw to match our publisher.

These are the steps we are planning on doing at the beginning of next week: 

- Try running the algorithm on existing .bag data, with IMU information (http://www.frc.ri.cmu.edu/~jizhang03/Datasets/gates_oscillating_motion.bag)
- Dive deeper into the actual code to better understand the inner workings.
- Getting more familiar with /tf, so that the offset and rotation between the axes is correctly registered, even if this doesn't explain why we aren't getting good results at standstill. Is this what is meant by the following segment (from http://wiki.ros.org/loam_velodyne), or do we need to do a manual projection? 

> "The input messages from the Velodyne and IMU follow the convention of x- pointing to the front, y- pointing to the left, and z- pointing upward. The IMU messages should be projected to align with the Velodyne frame before sending in."

Please stay in touch, and hopefully we can solve this together!

w4rlock999, 2018-10-06:
@luhcforgh looking forward for this. I am actually working on integratin lidar with IMU too, and facing problem like this!

BrunoEduardoCSantos, 2018-10-11:
@luhcforgh  Have you measured the performance on Kitti Dataset?

w4rlock999, 2018-10-12:
@luhcforgh what dataset did you use that contain imu data to test the lego loam? Can i have it to test it ?

luhcforgh, 2018-10-15:
@BrunoEduardoCSantos I'm unsure whether we actually did that, and we have since moved on to LeGO-LOAM since loop closure is a big perk for our application. 

@w4rlock999 We tried these datasets, which include IMU data. Don't forget to set the flag indicating that you are running bagged data, as described at the bottom of the LeGO-LOAM readme. https://github.com/RobustFieldAutonomyLab/jackal_dataset_20170608 

If you want to continue the discussion regarding LeGO-LOAM and IMU data, we have an active thread here instead: 
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/19

w4rlock999, 2018-10-20:
update for all, i actually near to success in using loam_velodyne with IMU, but not with laboshinl's package, but daobilige-su 's package.

the only problem is that the yaw, rotation on vertical axis, is really off, i think it's the problem with the pointcloud movement estimation, but the other rotation an translation is OK.

zorosmith, 2018-12-21:
@w4rlock999 How is your work going? Plz share your results or problems when integrating loam_velodyne with IMU