laboshinl/loam_velodyne #7: scanRegistration error

ctwu0314, 2016-08-08:
@laboshinl
I have successfully implemented according to your steps.
But when I connect the VLP-16 Lidar (roslaunch velodyne_pointcloud VLP16_points.launch  _ip_address:=192.168.1.201),
Usually happen the following error,

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
[scanRegistration-2] process has died [pid 13615, exit code -11, cmd /home/ctwu/ROS_Code/catkin_loam/devel/lib/loam_velodyne/scanRegistration __name:=scanRegistration __log:=/home/ctwu/.ros/log/1f7faff2-5d33-11e6-98e2-74d02b61e376/scanRegistration-2.log].
log file: /home/ctwu/.ros/log/1f7faff2-5d33-11e6-98e2-74d02b61e376/scanRegistration-2*.log
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
(the scanRegistration-2*.log can't find in the folder)

closed 2016-08-08 by ctwu0314

brianshu, 2017-02-04:
so how u solve this i met the similar issue. I previously got the connection part successfully worked, but after i try with google cartographer then got the same error as you do

AbnerCSZ, 2018-01-03:
I encountered the same problem, may you solve it? @ctwu0314   @brianshu  @laboshinl I suspect that the downsampled part has gone wrong(old version).

BrunoEduardoCSantos, 2018-07-16:
How do you guys solve the problem? I still have this pid issue

wangliuliu, 2018-10-17:
There are some related [issue](https://github.com/laboshinl/loam_velodyne/issues/71)
My pcl lib version is 1.7, Ubuntu 16.04.
I uninstall  pcl in source , and install by :
add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
apt-get update
apt-get install libpcl-*

cslxiao, 2018-12-04:
I build pcl in source and reinstall it, then it works.

AravindChandradoss, 2019-05-18:
@wangliuliu @cslxiao what is Ubuntu distribution? I am getting the following error.
```
Hit:6 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu bionic InRelease
Ign:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu bionic InRelease
Err:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.83 80]
Hit:9 http://archive.ubuntu.com/ubuntu bionic InRelease
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
did you guys use "utopic" distribution? 

Thanks

wuguandong, 2020-07-17:
Can try to comment out "add_definitions( -march=native )" in CMakeLists.txt.

mdq199668, 2020-08-25:
> There are some related [issue](https://github.com/laboshinl/loam_velodyne/issues/71)
> My pcl lib version is 1.7, Ubuntu 16.04.
> I uninstall pcl in source , and install by :
> add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
> apt-get update
> apt-get install libpcl-*

兄弟，我是用的手动安装编译，但还是解决不了这问题 请问你解决了吗