laboshinl/loam_velodyne #14: Error in roslaunch velodyne_pointcloud

kmotillo, 2017-04-28:
lacho@ubuntu:~/catkin_ws$ roslaunch velodyne_pointcloud VLP16_points.launch pcap:="~/catkin_ws/data.pcap"
... logging to /home/pelacho/.ros/log/fec73608-2c2c-11e7-8c13-000c29e7372c/roslaunch-ubuntu-52119.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:33811/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /velodyne_nodelet_manager_cloud/calibration: /home/pelacho/cat...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: ~/catkin_ws/data....
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [52131]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fec73608-2c2c-11e7-8c13-000c29e7372c
process[rosout-1]: started with pid [52144]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [52151]
process[velodyne_nodelet_manager_driver-3]: started with pid [52158]
process[velodyne_nodelet_manager_cloud-4]: started with pid [52163]
**[FATAL] [1493395744.252682183]: Error opening Velodyne socket dump file.
[FATAL] [1493395744.663492067]: Failed to load nodelet '/velodyne_nodelet_manager_driver` of type `velodyne_driver/DriverNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-2] process has died [pid 52151, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/pelacho/.ros/log/fec73608-2c2c-11e7-8c13-000c29e7372c/velodyne_nodelet_manager-2.log].
log file: /home/pelacho/.ros/log/fec73608-2c2c-11e7-8c13-000c29e7372c/velodyne_nodelet_manager-2*.log
[velodyne_nodelet_manager_driver-3] process has died [pid 52158, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load velodyne_driver/DriverNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_driver __log:=/home/pelacho/.ros/log/fec73608-2c2c-11e7-8c13-000c29e7372c/velodyne_nodelet_manager_driver-3.log].**
log file: /home/pelacho/.ros/log/fec73608-2c2c-11e7-8c13-000c29e7372c/velodyne_nodelet_manager_driver-3*.log
[velodyne_nodelet_manager_cloud-4] process has finished cleanly
log file: /home/pelacho/.ros/log/fec73608-2c2c-11e7-8c13-000c29e7372c/velodyne_nodelet_manager_cloud-4*.log

kmotillo, 2017-04-30:
I already solved this problem with the information from another issue, With a ROS launch script, it's necessary to provide the full path name to the PCAP file. example. 
/home/user/catkin_ws/file.pcap   correct
~/catkin_ws/file.pcap                 incorrect

AkshaySDesai, 2018-09-07:
Can you please share the link to the original issue that helped you solve the problem?

hierarch41, 2018-11-04:
I run：roslaunch velodyne_pointcloud VLP16_points.launch 
results：
pcap:="/home/laboshinl/Downloads/velodyne.pcap"
[VLP16_points.launch] is neither a launch file in package [velodyne_pointcloud] nor is [velodyne_pointcloud] a launch file name
The traceback for the exception was written to the log file

Can anyone tell me why this is?

fawadahm, 2019-07-15:
You would need to install the Velodyne driver for ROS.

sachinkm308, 2020-06-02:
> You would need to install the Velodyne driver for ROS.
Thanks, this worked!