laboshinl/loam_velodyne #16: How can I use LOAM with Velodyne64

lowmee, 2017-07-05:
Hallow,I'm using LOAM now,it's so great! I have modified code with "modified on linux" and "add velodyne HDL64E support" The procedure has compiled successfully. But when I execute launch there is only one node(transformMaintenance) live.The terminal displays as follows:

[laserOdometry-3] restarting process process[laserOdometry-3]: started with pid [14995] [laserOdometry-3] process has died [pid 14995, exit code -11, cmd /home/rmy/loam64/devel/lib/loam_velodyne/laserOdometry __name:=laserOdometry __log:=/home/rmy/.ros/log/038a2858-612a-11e7-b1f8-002324502a96/laserOdometry-3.log]. log file: /home/rmy/.ros/log/038a2858-612a-11e7-b1f8-002324502a96/laserOdometry-3*.log

I don't know why,can you help me? please

vkee, 2017-08-16:
Have you resolved this? I'm having issues too.

litgm, 2017-08-24:
#10 contains a has a suggestion by @ClementLeBihan

jianrui1, 2018-12-10:
@lowmee  I am working towards this.I got terrible result. can you  share some experience to me.