laboshinl/loam_velodyne #17: Loop Closure

athtest800, 2017-07-05:
Hello!
What is the process of loop closure the point cloud?
Do you use external loop closure technique?
The result has small drift from start to end.
Thank you.

vkee, 2017-08-11:
LOAM does not perform loop closures as stated in the paper (http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.448.434&rep=rep1&type=pdf), "Since our method is intended to push issues related to minimizing drift in odometry estimation, it currently does not involve loop closure."

xiaozhuka, 2017-09-22:
without loop closure my result is so bad
[image]