laboshinl/loam_velodyne #18: Array index size probrem in scanRegistration.cpp

norihiro-ito, 2017-09-22:
Hi, 
I found array size probrem in scanRegistration.cpp.
In the code, 4 arras has literal value 40000 was defined.
But VLP16's specs are horizontal resolution 0.1-04 degee and 16 vertical 16 channels.
So I think max array size needs 3600 x 16 = 57000 at least.
Actually, It is often over 40000 (48000 nearly.), and memory broken with scanRegistration process died.

thanks.

StefanGlaser, 2017-11-24:
Hi,
the static initialization of the arrays and number of scan rings is a rather unfortunate choice. I've looked into the code and noticed that most of the static arrays are only needed while working on an individual ring or ring region. I managed to adapt the implementation accordingly (in my fork) and it should now be independent of the used Velodyne lidar sensor. All internal buffers are resized accordingly during execution.
I still need to test it with a 64 beam lidar, but I'm quite confident that it'll work fine.

alittleharry, 2017-12-05:
In fact VLP16's  horizontal resolution is 0.2 degee.
Axactly the array size is not computed by 360/0.2, it's hard to explain  in one sentence. But i can tell you this:
for every ring, the number of points would below 2000;
for the whole scan, the number of points would below 30000; 
for every seconds, the number of points would below 300000.

to know why, you should to read the code of the velodyne driver.

norihiro-ito, 2017-12-05:
harry, thanks for your imformation.
However, when I played VLC16 bag data with debug print, the index was 48000 nearly.
and then scanregistration was aborted.
So i changed static value to 60000, and I could play the data normally.
Whats wrong? Now, my local code uses 60000 and looks good.

Thanks.

StefanGlaser, 2017-12-05:
Hi @norihiro-ito in recent commits I've replaced basically all static sized arrays with dynamic ones. If everything works as expected, you should not run into crashes with the current master version any longer.

norihiro-ito, 2017-12-06:
＠StefanGlaser, thanks for your great work.
Its almost totally changed.
I confirmed new code runs well with my bag data was crashed before.

Thank you.

norihiro-ito, 2017-12-06:
my issue is closed.

closed 2017-12-06 by norihiro-ito