laboshinl/loam_velodyne #21: A question about ShiftToStartIMU

xiesc, 2017-11-24:
I think the first line in this function is to calculate the distance shift from the first point to the current point in a scan period. So it should be  imuShiftFromStartXCur = imuShiftXCur - imuShiftXStart. But there is a  imuVeloXStart * pointTime behind them. I am confused about this.

StefanGlaser, 2017-12-04:
Hi, I'm not 100% sure on this one, but I would guess that the equation:
```
imuShiftFromStartXCur = imuShiftXCur - imuShiftXStart - imuVeloXStart * pointTime;
```
should be read as:
```
imuShiftFromStartXCur = imuShiftXCur - (imuShiftXStart + imuVeloXStart * pointTime);
```
where the idea is to calculate the shift (difference) in the actual, accumulated IMU position to the interpolated IMU position (using the start position and velocity).

There can be multiple IMU measurements per sweep, but the algorithm assumes linear motion within a sweep. I assume the calculation above is meant to achieve this for the IMU position. However, I'm not sure why only for the position and not the orientation.

xiesc, 2017-12-05:
Thanks for your answer! I think you are right and I found this part in Zhang's paper: In Chaption 7, B. Assistance from an IMU. In this part Zhang expained why only velocity but not orientation.
And i'm still confusing with `void luginIMURotation()`, in the `laserOdometry`. I thought it is used for accumulating the orientation of IMU to the laser odometry. Because the points used in laserOdometry have been transformed by using IMU orientation before.

alittleharry, 2017-12-05:
@StefanGlaser @xiesc i don't agree with you both. 
I must point out that the code implement is not exactly according to the paper content, in the paper the author use a 2D lidar with a motor to genrate 3D pointcloud. In scanRegistration.cpp , we use acc from IMU to acumulate the IMU odometry position and velocity, however the position and velocity will shift very fast! ( imagine that after a short time. the current accumulated  velocity have arrive 1000m/s ). So if use the  the right  calculation in theory(  imuShiftFromStartXCur = imuShiftXCur - imuShiftXStart), the distance between start and end of a frame is very large which would made the point pose compensation worse.
For orientation , it different from position because it is estimated in other place and not accumulated here.  According my experence with IMU, the orientation estimeted using EKF  is very stable and error not accumulated.
as for:
 imuShiftFromStartXCur = imuShiftXCur - imuShiftXStart - imuVeloXStart * pointTime;
here in my opinion here the author asume the speed of start to be zero, so the equation  remove shift caused by shifted speed of start time. The  equation is not right in theory but works better.

StefanGlaser, 2017-12-05:
@alittleharry thats an interesting point. Thanks for clarifying. I was actually wondering where the IMU position / velocity drift compensation is incorporated after looking at the IMU accumulation logic. This also explains why it's only done for position and not orientation.  
@xiesc sorry for the confusion.

xiesc, 2017-12-06:
@StefanGlaser  I think that your first comment was right. As for IMU, the accX,Y,Z, will vibrate fast but not velocity. velocity is accumulated by acc. but in a very short time step, the velocity reaching 1000m/s as @alittleharry  seems impossible. And taking a look at Fig 12a  in Zhang's paper, the trajectory of IMU is compromised. And in his paper he explained this part. like: 
We attach an Xsens MTi-10 IMU to the lidar to deal with
fast velocity changes. The point cloud is preprocessed in
two ways before sending to the proposed method, 1) with
orientation from the IMU, the point cloud received in one
sweep is rotated to align with the initial orientation of the lidar
in that sweep, 2) with acceleration measurement, the motion
distortion is partially removed as if the lidar moves at a const
velocity during the sweep. The point cloud is then processed
by the lidar odometry and mapping programs

StefanGlaser, 2017-12-09:
@xiesc well, let's take a closer look...

If my first comment was correct, what would a calculation look like? First, there needs to be some sort of calculation of the average speed during a sweep - e.g. ```(end_vel + start_vel) / 2``` - or simply take one velocity vector at the start or end of a sweep. Once we decided for an average speed, an interpolation would look like ```pos_shift = start_pos + avg_vel * deltaT```. However, I can't find neither of these two calculations within the code.

Just out of curiosity for the actual values, I printed the accumulated velocities of the IMU for the "gates_oscillating_motion.bag" (which contains IMU data). And after almost one minute, the accumulated velocity is around 23 m/s (> 80km/h) in a situation where the robot came to a full stop. Using the 23 m/s for transforming the incoming point cloud in such a situation would have a serious impact on the result. On the other hand, if the accumulated IMU state and the start IMU state have similar velocities, they would simply cancel out, resulting in a zero speed assumption as mentioned above.

Looking at the calculations involved, we first calculate this ```imuShiftFromStart``` vector and then directly use it to shift the point position. Within this context, the explanation of @alittleharry makes sense to me. Especially considering that there is no general IMU position / velocity drift compensation.

As far as I understand, the laser odometry component will apply the (constant) distortion correction after it optimized the current odometry estimation. In case there is no IMU data available, there is no straight forward option (within scan registration) to correct the incoming point cloud for distortion. Using the above zero-speed assumption, the distortion correction is relatively similar to having no IMU at all (as long as the velocity is constant during a sweep). However, in case of fast velocity changes within two subsequent IMU states, the ```imuShiftFromStart``` vector would reflect this velocity change and compensate for it in the point cloud. As the laser odometry assumes constant speed during a sweep, this approach actually compensates varying speeds within a sweep, before the constant speed assumption is applied. Pretty clever actually if I think twice about it...

Sorry for the long answer, but I hope that it helps clarifying open questions.

2amour, 2018-04-01:
@StefanGlaser 
What is the function of the void PluginRatation()?
Thanks in advance!

gloomysunday, 2019-07-21:
@xiesc @2amour 
The motion in one sweep is separated into two parts. 

Firstly, orientation from IMU and nonlinear motion from acceleration are processed by **BasicScanRegistration::projectPointToStartOfSweep** (the code, _imuPositionShift = _imuCur.position - _imuStart.position - _imuStart.velocity * relSweepTime, is mean nonlinear motion from acceleration).  

Secondly, the orientation drift and linear motion (constant velocity), noted as **_transform** in class BasicLaserOdometry, are processed by lidar odometry. 

So, after the optimization of lidar odometry, the global odometry pose transformation **_transformSum** is obtained by accumulating orientation drift and linear motion as **BasicScanRegistration::accumulateRotation** , then accumulating orientation from IMU and nonlinear motion as **BasicScanRegistration::pluginIMURotation**.

closed 2021-11-30 by xiesc