laboshinl/loam_velodyne #36: Great performance! But potential bug?

mickeys11375, 2017-12-09:
Hi Folks,

Thank you so much for make contributions to the LOAM code! I tested the algorithm using my own data set and observed no drift at all even after 1000 meters travel in urban area with elevation changes.

[image]

mickeys11375, 2017-12-09:
However, I think there **might** be a bug in the scanRegistration.cpp. (I am not sure)
When LOAM uses IMU to adjust the point cloud using the following code:
`while (imuPointerFront != imuPointerLast) {
        if (timeScanCur + pointTime < imuTime[imuPointerFront]) {
          break;
        }
        imuPointerFront = (imuPointerFront + 1) % imuQueLength;
      }

      if (timeScanCur + pointTime > imuTime[imuPointerFront]) {
        .....................
      } else {
        .....................
      }
      if (i == 0) {
        imuRollStart = imuRollCur;
        imuPitchStart = imuPitchCur;
        imuYawStart = imuYawCur;
        .....................
      } else {
        ShiftToStartIMU(pointTime);
        VeloToStartIMU();
        TransformToStartIMU(&point);
      }`
I notice that the _timeScanCur_ is later than the latest _imuTime[imuPointerLast]_. This means that `if (timeScanCur + pointTime > imuTime[imuPointerFront])` will always be true. So the interpolation is not used. Should the _timeScanCur_ be the starting time of a scan or the ending time of a scan? Maybe the starting time? Then that if code will make sense.

StefanGlaser, 2017-12-10:
That's a nice result. Thanks for sharing.

The ```timeScanCur``` is used as start time of the sweep. It holds the time stamp of the header of the currently processed point cloud message.

IMU measurements are typically received in a much higher frequency than point clouds and should therefore be later than at least the start time of a scan. If this is not the case, then either the IMU frequency is too low, or the driver processing the point clouds is assigning "wrong" / the current time as time stamps to the point cloud messages.

Maybe try to subtract the ```scanPeriod``` from the incoming message time stamp and check if the resulting behavior (and mapping performance) is more consistent with your expectations.

Another potential problem is the interpolation of the point times, as it sometimes interpolates based on a wrong range. As a result, the scan time interpolation factor is sometimes less than zero or grater than one. However, in such a case only some of the points should be later than the latest IMU measurement - certainly not all.

mickeys11375, 2017-12-10:
@StefanGlaser 
Thanks for the explanation. What you said makes perfect sense. I think there might be some issues in the velodyne driver node.

closed 2017-12-10 by mickeys11375

deepmeng, 2017-12-16:
@mickeys11375 Congratulations for your great result! I am just wondering about how you save the map built by loam? And how to  build such a great map? What msgs you used? Only pointcloud and IMU?