laboshinl/loam_velodyne #39: How to build a big outdoor map?

deepmeng, 2017-12-19:
Does anyone meets the problem when build a big outdoor map? When I build a map with a size of 3km*3km, loam can only build a fixed-size map and this map less than actual size of map. Where can I modify the code to  make it?

TixiaoShan, 2017-12-19:
```
for (int i = centerCubeI - 2; i <= centerCubeI + 2; i++) {
    for (int j = centerCubeJ - 2; j <= centerCubeJ + 2; j++) {
      for (int k = centerCubeK - 2; k <= centerCubeK + 2; k++) {
      }
    }
}
```
This shows the surrounding map. Make the number larger, you can visualize more map. But it will slows the system down since this is also the map sent to kdtree. You can write your own code to visualize it to save some computational resources.

TixiaoShan, 2017-12-19:
```
_laserCloudWidth(21),
_laserCloudHeight(11),
_laserCloudDepth(21),
```

These 3 numbers show all the cubes that save the map. Each cude is 50m*50m*50m. Increase the number will increase the size of the map.

TixiaoShan, 2017-12-19:
To save the point cloud, you can read the tutorial [here](http://pointclouds.org/documentation/tutorials/writing_pcd.php).

deepmeng, 2017-12-22:
Thanks very much! I tried it by following your advice. But it does not work.

closed 2017-12-22 by deepmeng