laboshinl/loam_velodyne #40: [laserMapping-3] process has died [pid 29080, exit code -11

kentzou, 2018-01-23:
@dimatura@evenator @laboshinl @facontidavide 
[image]

I run the loam_velodyne according to the github page.
In first terminal
$ cd ~/catkin_ws/src/
$ git clone https://github.com/laboshinl/loam_velodyne.git
$ cd ~/catkin_ws
$ catkin_make -DCMAKE_BUILD_TYPE=Release
$ source ~/catkin_ws/devel/setup.bash
In second terminal
$ roslaunch loam_velodyne loam_velodyne.launch
$ rosbag play ~/Downloads/velodyne.bag 

and then ,show the follow error:
`[laserMapping-3] process has died [pid 29080, exit code -11, cmd /home/yong/catkin_ws/devel/lib/loam_velodyne/laserMapping __name:=laserMapping __log:=/home/yong/.ros/log/caae77aa-ffda-11e7-b7e2-1002b5cc091b/laserMapping-3.log].
log file: /home/yong/.ros/log/caae77aa-ffda-11e7-b7e2-1002b5cc091b/laserMapping-3*.log`

I use ubuntu14.04 and PCL 1.8.
I found it caused by this code in LaserMapping.cpp:
`matX = matAtA.colPivHouseholderQr().solve(matAtB);`
Could you tell me how to solve it.

ClementLeBihan, 2018-02-12:
Hi, got the same problem. Waiting for advices ...

facontidavide, 2018-07-05:
@laboshinl should take a look at it, but since it is somehow related to a change I submitted, I will try to take a look at it.
Can anyone share a rosbag to reproduce the error, please?

BrunoEduardoCSantos, 2018-07-05:
Hey,
How can I share the rosbag? Github doesnt allow it.

BrunoEduardoCSantos, 2018-07-05:
The error is:

> [laserMapping-8] process has died [pid 2688, exit code -6, cmd /home/bruno/catkin_ws/devel/lib/loam_velodyne/laserMapping __name:=laserMapping __log:=/home/bruno/.ros/log/00585b14-802d-11e8-8f50-2816ad8422ea/laserMapping-8.log].
log file: /home/bruno/.ros/log/00585b14-802d-11e8-8f50-2816ad8422ea/laserMapping-8*.log
[multiScanRegistration-6] process has died [pid 2662, exit code -11, cmd /home/bruno/catkin_ws/devel/lib/loam_velodyne/multiScanRegistration /multi_scan_points:=/velodyne_points /imu/data:=/raw_imu __name:=multiScanRegistration __log:=/home/bruno/.ros/log/00585b14-802d-11e8-8f50-2816ad8422ea/multiScanRegistration-6.log].
log file: /home/bruno/.ros/log/00585b14-802d-11e8-8f50-2816ad8422ea/multiScanRegistration-6*.log

facontidavide, 2018-07-05:
@BrunoEduardoCSantos [let me google it for you](https://www.google.com/search?client=ubuntu&channel=fs&q=how+to+share+a+file+on+internet&ie=utf-8&oe=utf-8)

BrunoEduardoCSantos, 2018-07-05:
Any dataset available it has the error.

facontidavide, 2018-07-05:
in that case, I am not able to reproduce the error. @laboshinl any idea?

BrunoEduardoCSantos, 2018-07-05:
Which version do you have?

BrunoEduardoCSantos, 2018-07-06:
So, I run on gdb mode and no error comes up.

facontidavide, 2018-07-06:
I REALLY don't want to sound rude, so don't take me wrong, __but__:

1) This is @laboshinl 's project, not mine. But he has been silent so far. I am not an official nor extra-official maintainer of this repository.
2) Nevertheless, I will be happy, just for fun, to take a look at this issue, __IF__ you or anyone else sends me a rosbag to reproduce the error. Otherwise, I will not spend a single minute on it.

You can use my email: davide.faconti@gmail.com if you don't want to post the link here.

Cheers

Williamc17, 2018-07-27:
Got the same error. I had sent  you@facontidavide  the rosbag. Waiting your answer.

facontidavide, 2018-07-27:
Thanks @Williamc17 . I will take a look when I have time. Don't hold your breath ;)

Jackey-Huo, 2018-08-11:
@kentzou @facontidavide  I have met exactly the same problem with you. After some debug and search time, I find it caused by the mismatch compiler flag of prebuild pcl library. I reinstall pcl, build it from source, then everything is Okey.  

Related issue and pr from pcl library is here: [issue2364](https://github.com/PointCloudLibrary/pcl/issues/2364) , [pr2100](https://github.com/PointCloudLibrary/pcl/pull/2100)

Hope it could be helpful

Allen0, 2018-09-17:
Jackey is right, update pcl to version 1.8, the problem is fixed!!
Thx man

YangSiri, 2019-12-13:
Hi, @kentzou 

Recently i run into similar situation like you not only in _loam_ but also _A-LOAM_, _loam_livox_. It is all about the laserMapping process. Can you tell me your solution to this ?

Looking forward to your generous help ;)

liunao123, 2021-07-31:
Comment out 35th row in CMakeLists.txt, like below.
#add_definitions( -march=native )
then catkin_make again, everything is OK in my PC.