laboshinl/loam_velodyne #41: How to show the complete map in Rviz?

RuochenYin, 2018-02-05:
I want to build a big outdoor map by using the Loam. But in the rviz, it just show a fixed-size map. Actually all the points cloud still exist in the rviz but  it just show the sub map around the car. So how to show the complete map? Thanks.

StefanGlaser, 2018-02-06:
Your case sounds very similar to #39 however, the answers somehow disappeared...  
@deepmeng maybe you're also interested, or may help on this one.

The mapping component of loam uses a sliding window approach for holding the accumulated point cloud map around the robot. At the beginning, the current implementation initializes a 3D grid around the initial position. Each cube within this grid is represented by a point cloud.  
For each incoming sweep, the mapping component then only updates a small region of cubes around its current position. The updated region is then concatenated into one big point cloud, downsampled and published via the ```laser_cloud_surround``` topic.  
The robot can move freely within the grid. However, as soon as it approaches any boundary of the grid, the grid is shifted in the corresponding direction, loosing previously accumulated map cubes on the opposite side of the grid. The size of the grid is currently hard-coded within the ```LaserMapping``` constructor (```_laserCloudWidth```, ```_laserCloudHeight```, ```_laserCloudDepth```). The size of each cube is (also hard-coded) 50x50x50 meter.

You should be able to publish the complete map to the ```laser_cloud_surround``` topic by simply concatenating all cubes into one big point cloud, instead of only the updated region. You'll find the corresponding code at the top of the ```LaserMapping::publishResult()``` method. Something like:
```cpp
// accumulate full map cloud
_laserCloudSurround->clear();
for (int i = 0; i < _laserCloudNum; i++) {
  *_laserCloudSurround += *_laserCloudCornerArray[i];
  *_laserCloudSurround += *_laserCloudSurfArray[i];
}
```
should do the work. However, I guess this will have some negative impact on the runtime of the component, as it involves a lot of computation just to be able to display the whole map in the UI in real time.

For saving the full map into a .PCD file, you can do the same as above (concatenate all point cloud cubes into one big point cloud) and save the resulting point cloud instance using the PCL. Of course, I strongly recommend to implement an external trigger for saving the map into a .PCD file, as it's not reasonable to do this within the main loop.

RuochenYin, 2018-02-07:
@StefanGlaser Thanks a lot, it really helps. And I've been in contact with deepmeng.

closed 2018-02-07 by RuochenYin

chris0731, 2019-07-15:
@Rolanding  can you please share what you learnt from deepmeng?

Rmargin, 2020-07-10:
HaHa

hhongwei1009, 2024-06-25:
> Your case sounds very similar to #39 however, the answers somehow disappeared... @deepmeng maybe you're also interested, or may help on this one.
> 
> The mapping component of loam uses a sliding window approach for holding the accumulated point cloud map around the robot. At the beginning, the current implementation initializes a 3D grid around the initial position. Each cube within this grid is represented by a point cloud. For each incoming sweep, the mapping component then only updates a small region of cubes around its current position. The updated region is then concatenated into one big point cloud, downsampled and published via the `laser_cloud_surround` topic. The robot can move freely within the grid. However, as soon as it approaches any boundary of the grid, the grid is shifted in the corresponding direction, loosing previously accumulated map cubes on the opposite side of the grid. The size of the grid is currently hard-coded within the `LaserMapping` constructor (`_laserCloudWidth`, `_laserCloudHeight`, `_laserCloudDepth`). The size of each cube is (also hard-coded) 50x50x50 meter.
> 
> You should be able to publish the complete map to the `laser_cloud_surround` topic by simply concatenating all cubes into one big point cloud, instead of only the updated region. You'll find the corresponding code at the top of the `LaserMapping::publishResult()` method. Something like:
> 
> ```c++
> // accumulate full map cloud
> _laserCloudSurround->clear();
> for (int i = 0; i < _laserCloudNum; i++) {
>   *_laserCloudSurround += *_laserCloudCornerArray[i];
>   *_laserCloudSurround += *_laserCloudSurfArray[i];
> }
> ```
> 
> should do the work. However, I guess this will have some negative impact on the runtime of the component, as it involves a lot of computation just to be able to display the whole map in the UI in real time.
> 
> For saving the full map into a .PCD file, you can do the same as above (concatenate all point cloud cubes into one big point cloud) and save the resulting point cloud instance using the PCL. Of course, I strongly recommend to implement an external trigger for saving the map into a .PCD file, as it's not reasonable to do this within the main loop.

It didn't work.Is there a specific method?