laboshinl/loam_velodyne #44: Support for actuated 2D laser scanners

YoshuaNava, 2018-03-01:
Hi,
I'm working with an actuated 2D LIDAR, and I was wondering if it's within your plans to integrate the variants of LOAM that work with this type of sensor: [loam_back_and_forth](https://github.com/daobilige-su/loam_back_and_forth) and [loam_continuous](https://github.com/daobilige-su/loam_continuous).

If the answer is yes, I would be greatly interested in contributing.

Kind regards,
Yoshua

StefanGlaser, 2018-03-15:
Hi @YoshuaNava,  
I started a new implementation some months ago, combining all three approaches into one project. However, I kind of run out of time.

Combining all three implementations is pretty straight forward, you need to:
1. Define and publish partial sweeps, as a sweep in the back-and-forth / continuous case is arriving sequentially
2. Process partial sweeps in the odometry component (optimize odometry during a sweep)
3. Only forward full sweeps to mapping component
4. Handle variable sweep times
5. Adapt some parameters

Apart from that, the back-and-forth / continuous implementations uses XYZHSV points for storing additional information about a point, whereas loam_velodyne encodes the relative scan time and scan ring/number within the intensity of an XYZI point.

All the best and good luck!

YoshuaNava, 2018-03-22:
Hi @StefanGlaser,
Thank you for the advice! I started the integration of 2D actuated scanners. I hope I can have it ready soon, to make a pull request.