laboshinl/loam_velodyne #47: Export trajectory

litgm, 2018-03-14:
Hi
I want to use LOAM for it's odometry output.

So far I've tried to record the three odomertry topics:
- aft_mapped_to_init
- laser_odom_to_init
- integrated_to_init
using "rostopic", but the timestamps seems off

did anyone tried to use these topics?

AhmedAboutaleb94, 2019-01-08:
Have you resolved this issue ??

zahirulislamovich, 2019-02-03:
Hi. Thanks all to this forum. Can any give me process about taking 200 hz imu, trajectory export and gps time stamp

Thanks again

closed 2019-02-03 by litaws

michaelczhou, 2019-02-16:
> Hi
> I want to use LOAM for it's odometry output.
> 
> So far I've tried to record the three odomertry topics:
> 
> * aft_mapped_to_init
> * laser_odom_to_init
> * integrated_to_init
>   using "rostopic", but the timestamps seems off
> 
> did anyone tried to use these topics?

so how did you do this?thx

litaws, 2019-02-20:
you can dump a topic into a file (see the `rostopic` documentation) and then parse it. 
I cannot share code I used for parsing but it is rather straightforward.

zahirulislamovich, 2019-02-20:
Dear litwas. I have searched through all over internet but i need straight forward code. I think its so easy asbits find to be so difficult for me... Will be very helpfull if anyone can provide me.. Thanks in advance. Or at least any clue.

claydergc, 2019-02-20:
@zahirulislamovich here there is an example https://github.com/claydergc/loam_velodyne_kitti_ros/blob/master/scripts/record_loam_odometry.py

zahirulislamovich, 2019-02-20:
Thank you. So great