laboshinl/loam_velodyne #51: when the distance is longer than 500m,does it work?

xbcdbc, 2018-03-27:


StefanGlaser, 2018-04-29:
Hi,  
the current implementation will store a maximum area of 1050 x 1050 x 550 meter (x, y, z) if I remember correctly. This area is implemented as a sliding window, moving with the robot if it exceeds one of the boundaries.

Apart from that, during the registration and mapping of a new sweep, only a small sub-area of 250 x 250 x 250 meter around the robot is used.

There is no restriction on the overall distance the robot can travel.

BrunoEduardoCSantos, 2018-06-27:
Hi,
Maybe is a related issue. I built a map using odometry, although there is a scale problem compared with GPS. The LOAM algorithm shorts down the maximum value in the axis, even though the mapping is correct. Any ideas why?