laboshinl/loam_velodyne #53: [pcl::VoxelGrid::apply Filter] Leaf size is too small for the input dataset.

Xiangzhaohong, 2018-04-10:
Did anyone meet this problem? I use loam with HDL64.pacp dataset and can't solve it.

elahea2020, 2018-08-20:
same, would highly appreciate it if someone could tell me how to resolve this issue. 
Thanks

Shantnu12345, 2018-11-10:
I also have the same error. Maybe this is coming from PCL. It would be nice to know if someone has resolved it.

Thanks,
Shantnu

zakwan99, 2018-11-19:
anyone has solved it??

Yeah2333, 2018-12-03:
there maybe some error with your vlp-16 config.i change rpm to defalut 600 .the error has been solved. but when i change rpm to 300. i have this error about leaf size. to change vlp-16 config ,you should connect it and open chorme or firefox access 192.168.1.201 to change your rpm