laboshinl/loam_velodyne #60: Online SLAM using VLP-16 LiDAR

ekow-mensah, 2018-05-14:
Hi all, I am trying to implement online 3D SLAM using the VLP-16 LiDAR how to I do this with LOAM without having to use rosbags or pcap files?

julianst, 2018-05-18:
It should also work with live data, as it just subscribes to the same topic as it does when using rosbags. All you need to do is publish your point cloud as a rod topic. A tutorial for doing so can be found here: http://wiki.ros.org/velodyne

ekow-mensah, 2018-05-20:
okay thank you. I have tried this and it works but I do not understand what is the purpose of the red arrows?

imawesome61, 2018-11-17:
hello monsieur, it would bevery kind and humble of you if you can help me create a cloud point for .pcap format data captured from vlp16 lidar velodyne sensor .seeking forward for your reply. Thank you in advance.