laboshinl/loam_velodyne #62: Drift in Z-Direction

julianst, 2018-05-16:
As also observed by @CansenJIANG @TopGunSnake @kamiyuanyang @nikitaporje the resulting map and position is drifting in Z-direction, which results in a bended map:
[image]
[image]

julianst, 2018-05-24:
I was able to confirm tis issue with the VLP 16 sample pcaps 2014-11-10-10-36-54_Velodyne-VLP_10Hz-County Fair.pcap and 2014-11-10-10-48-31_Velodyne-VLP_20Hz_Hecker Pass.pcap from https://midas3.kitware.com/midas/folder/12979 as well as live VLP 16 data running Ubuntu 16.04 LTS and ROS kinetic.

Is there anyone having the same issue?

closed 2018-06-11 by julianst

ashnarayan13, 2018-06-21:
Hi, 

What was the fix for this problem? 
When I run loam with a direct LiDAR stream, I get this drift as well. This is more apparent when the sensor is stationary. 

Thanks

kongcheng08, 2018-08-21:
I have the same problem  with vlp16 LiDAR when running outdoor environment , Now have any solution ?

zr2539168, 2018-09-28:
I face the same problem, did you fix it? Thanks

mistercherryy, 2023-06-09:
Anyone solved this issue or has a clue how to solve it? I am suffering the same issue.