laboshinl/loam_velodyne #63: Detach LOAM from ROS

YoshuaNava, 2018-05-21:
Hi,
I think that it could be extremely useful to detach LOAM from ROS, such that:

1. It can be used as a stand-alone method for motion estimation, e.g. by instantiating a class "LoamWorker", which is feed with point clouds and IMU messages. It could return odometry, mapping and integrated motion estimates, as well as point clouds that contain features, the map and the current cloud registered in the map.
2. As a consequence of 1, be able to tune the parameters of the algorithm to increase accuracy.
3. Also because of 1, be able to integrate Loam into more complex systems, non dependent on ROS.
4. Make it easier to run tests with KITTI or other datasets.

There was substantial work in this direction from users @adrelino and @sebdi, contained in [this repo](https://github.com/adrelino/loam_velodyne-1). I'm also currently working on this for my thesis, but it may take some time until I submit a pull request.

What do you think? Would anyone be interested in collaborating for achieving this?

Kind regards,
Yoshua Nava

jlblancoc, 2018-05-21:
I absolutely agree with this approach, not only for this algorithm but for
every single Ros package!

But I don't have spare time to help with this package, at least not in the
short term.

Cheers

adishavit, 2018-05-21:
I did this recently. 
I plan to submit a PR for this soon.

YoshuaNava, 2018-05-21:
@jlblancoc Thank you! Hope we can take this initiative forward

@adishavit That's awesome!
Could I ask you some questions about what you did? I have tried to phrase them so that they can be answered with in a brief way. Now, I assume you took away all references to ROS present in the registration, odometry, mapping and transform maintenance classes. Then,

1. Did you create individual wrappers for each module, or one wrapper for LOAM as a whole?
2. Did you fully replace the message passing (via ROS) between the registration, odometry mapping and transform maintenance modules?
3. Assuming the answer to 2 is a yes (because you're running LOAM on Windows), how did you replicate the effect of having the callback queues holding new messages (from the IMU, odometry, transforms, etc)?
3. Did you observe any noticeable impact on the performance?

Thank you very much.

adishavit, 2018-05-22:
@YoshuaNava : 

1. I separated the ROS "transport" layer from each module from the underlying algorithm.
So for `ScanRegistration` I now have the algorithm class `BasicScanRegistration` which depends only on PCL, and which `ScanRegistration` derives from, though you could also use aggregation or `BasicScanRegistration` directly (no `protected` members and very few `virtual` functions).
This is done for each module and each is still built into an independent lib as before.
I kept the original interface as is (even where it is currently somewhat clunky and redundant).
2. The only ROS facility I needed to replace was using `std::chrono` instead of `ros::Time` there rest remains in the ROS-using transport layer.
3. The code does not have any tests: not unit tests nor performance and qualitative tests. Performance should be identical as I did not try to make any functional changes (up-to any bugs I may have introduced (or fixed)).

YoshuaNava, 2018-05-29:
Hey @adishavit,
That would be awesome! Please keep me updated

sume-cn, 2018-09-19:
@adishavit 
[this repo](https://github.com/claydergc/loam_velodyne_kitti) can be tansplanted to windows, I've tested it, but the output trajectory is deformed

claydergc, 2019-01-07:
@sume-cn the error in my repo is produced because I didn't add the time stamp to the messages and that's quite important. It would help if someone could do that.