laboshinl/loam_velodyne #71: loam_velodyne failed

mantangL, 2018-06-24:
I have built the loam project successfully.And when I launch in ros kinetic ,it's successful too. But when I play the bag it's error in my computer.such as:
 ... logging to /home/lmt/.ros/log/d9401fc2-77a2-11e8-a910-902b3433bbc4/roslaunch-lmt-8349.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lmt:33200/

SUMMARY
========

PARAMETERS
 * /laserMapping/scanPeriod: 0.1
 * /laserOdometry/scanPeriod: 0.1
 * /multiScanRegistration/lidar: VLP-16
 * /multiScanRegistration/scanPeriod: 0.1
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    laserMapping (loam_velodyne/laserMapping)
    laserOdometry (loam_velodyne/laserOdometry)
    multiScanRegistration (loam_velodyne/multiScanRegistration)
    rviz (rviz/rviz)
    transformMaintenance (loam_velodyne/transformMaintenance)

auto-starting new master
process[master]: started with pid [8360]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d9401fc2-77a2-11e8-a910-902b3433bbc4
process[rosout-1]: started with pid [8373]
started core service [/rosout]
process[multiScanRegistration-2]: started with pid [8382]
process[laserOdometry-3]: started with pid [8389]
process[laserMapping-4]: started with pid [8392]
process[transformMaintenance-5]: started with pid [8393]
process[rviz-6]: started with pid [8409]
[ INFO] [1529840195.272982324]: Set scanPeriod: 0.1
[ INFO] [1529840195.306128817]: Set  VLP-16  scan mapper.
0x1c04540 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1069d50) ): Attempt to set a screen on a child window.
0x1bfa9d0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1069d50) ): Attempt to set a screen on a child window.
0x1c02c00 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1069d50) ): Attempt to set a screen on a child window.
0x1c06060 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x1069d50) ): Attempt to set a screen on a child window.
[multiScanRegistration-2] process has died [pid 8382, exit code -11, cmd /home/lmt/loam_ws/devel/lib/loam_velodyne/multiScanRegistration /multi_scan_points:=/velodyne_points __name:=multiScanRegistration __log:=/home/lmt/.ros/log/d9401fc2-77a2-11e8-a910-902b3433bbc4/multiScanRegistration-2.log].
log file: /home/lmt/.ros/log/d9401fc2-77a2-11e8-a910-902b3433bbc4/multiScanRegistration-2*.log

And then I put this project in other computer It is successful. I can't find the reason of this failure.Why it is fail?

BrunoEduardoCSantos, 2018-07-16:
I have precisely the same issue.  Did you find any solution?

Jackey-Huo, 2018-08-08:
@mantangL would you mind provide your pcl lib version and g++ version? I encountered the same issue before, and find it related to pcl voxelgrid method

mantangL, 2018-08-15:
@Jackey-Huo  pcl lib version is 1.7      g++ version is 5.4.0

mantangL, 2018-08-15:
@Jackey-Huo  how do you solve it?

Jackey-Huo, 2018-08-26:
My pcl lib version is 1.8, g++ 7.2, Ubuntu 18.04.

I found that when I use the apt installed pcl lib, multiScanRegistration would crash immediately after I start play rosbag. After some search I found it's caused by the mismatch between pre-build pcl binary and native compiler options. So I removed the previous pcl lib and build it from source, Then everything works fine!

There are some related issue & pr from pcl community: [issue2013](https://github.com/PointCloudLibrary/pcl/issues/2013) , [PR 2100](https://github.com/PointCloudLibrary/pcl/pull/2100)

Hope you find it helpful :)

xuhaoda1995, 2019-01-11:
@Jackey-Huo I found i built pcl from source then noted that can't find pcl_conversionsConfig.cmake,can you give some detail about build pcl? thank you.

Jackey-Huo, 2019-01-12:
@xuhaoda1995 pcl_conversions is a ros package, providing pcl data structure conversion from pcl lib and ros. It should be provided by ros.

The following cmd may help you:
`locate pcl_conversionsConfig.cmake`

If no file found, you have to install the package. you can install it from apt by typing:
`sudo apt-get install ros-<your ros version>-pcl-conversions`

Hope it can help you ;)

xuhaoda1995, 2019-01-20:
@Jackey-Huo thank you for your help. I remove the pcl lib and want to build it by myself for reason of loam not running. Now I solve the problem by building pcl_conversions source.

slmat27, 2019-02-07:
@Jackey-Huo would you kindly advise on how exactly did you remove the pcl lib, my pcl lib version is 1.8, g++ 7.3, Ubuntu 18.04 and ros melodic and getting the same error. Thank you in advance!

Jackey-Huo, 2019-02-07:
1. Remove the apt installed pcl lib
`sudo apt-get remove --purge libpcl*`

2. Verify loam cannot find any pcl lib
`
cd path/to/your/catkin_ws
rm -rf build devel
catkin_make
`
If the build failed due to lack of pcl lib, then you have remove pcl successfully :)

liumengucas, 2019-02-21:
@slmat27 We met the same error with the same working environment. Do u have ever solved it now?

liumengucas, 2019-02-21:
We met the same error with the same working environment. Did you find any solution? I followed the above instructions, but it is still not working...

slmat27, 2019-02-21:
I solved it by installing pcl-1.9 from source, and everything is okay since then

You can download the latest release from here: https://github.com/PointCloudLibrary/pcl/releases

And the installation from source steps can be found here: http://www.pointclouds.org/documentation/tutorials/compiling_pcl_posix.php

liumengucas, 2019-02-22:
@slmat27 Thanks so much. The "multiScanRegistration" error can be avoided. However, I met the new error log "[laserMapping-4] process has died". I have to solve the new problem...

slmat27, 2019-02-22:
@liumengucas I would more information regarding the error to be able to support you.

AravindChandradoss, 2019-03-29:
Hey! @liumengucas I also got the same error? did you solve the issue?

Jackey-Huo, 2019-03-29:
@AravindChandradoss Would you mind to provide information about your cpu and OS? I'd like to know the gdb backtrace from crash point as well

AravindChandradoss, 2019-03-29:
Ubuntu 18.04, intel core i7
ROS melodic. 
Got this error! 

```
[laserMapping-4] process has died [pid 6162, exit code -11, cmd /home/aravinddoss/slam/loam/devel/lib/loam_velodyne/laserMapping __name:=laserMapping __log:=/home/aravinddoss/.ros/log/4b342834-5237-11e9-b0cb-34e6ad887d40/laserMapping-4.log].
log file: /home/aravinddoss/.ros/log/4b342834-5237-11e9-b0cb-34e6ad887d40/laserMapping-4*.log

```

Jackey-Huo, 2019-03-30:
> Ubuntu 18.04, intel core i7
> ROS melodic.
> Got this error!
>
> ```
> [laserMapping-4] process has died [pid 6162, exit code -11, cmd /home/aravinddoss/slam/loam/devel/lib/loam_velodyne/laserMapping [...]

@AravindChandradoss 

1. I need the detailed cpu info, please run the following command and paste output
    `lscpu`

2. The error just show laserMapping node crashed, we need deep debug info. Please build the whole project in debug mode ( you can change the compiler flags in CMakeLists.txt file ), and launch it with gdb.
Inside the gdb, run it till the crash point, and print backtrace of it.

the following links may help you finish it:
    https://bytefreaks.net/programming-2/cc-how-do-you-set-gdb-debug-flag-g-with-cmake
    https://sourceware.org/gdb/onlinedocs/gdb/Backtrace.html
    https://answers.ros.org/question/47210/how-to-roslaunch-node-in-gdb/

AravindChandradoss, 2019-03-31:
Hey! @Jackey-Huo 
I did the following, 
```
1) catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo
```
2) modified launch file, 
```
<node pkg="loam_velodyne" type="laserMapping" name="laserMapping" output="screen"> 
to 
<node pkg="loam_velodyne" type="laserMapping" name="laserMapping" output="screen" launch-prefix="xterm -e gdb --args">
```
3) tried launch
```
roslaunch loam_velodyne loam_velodyne.launch 
```
Got this error
```
RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

xterm -e gdb --args /home/aravinddoss/slam/loam/devel/lib/loam_velodyne/laserMapping __name:=laserMapping __log:=/home/aravinddoss/.ros/log/983c101e-5355-11e9-aabf-34e6ad887d40/laserMapping-4.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file

```
Am I missing anything?  

I also referred this [link](http://lsi.vc.ehu.es/pablogn/investig/ROS/9781782161448_Chapter-03.pdf
)
btw, Thanks!!!

Jackey-Huo, 2019-04-01:
@AravindChandradoss 
Please make sure you have installed xterm and gdb

AravindChandradoss, 2019-04-02:
I installed PCL.1.9.1 from source and tried to launch the file again. 
This time, the SLAM was working but only for a few seconds. Then I got a segmentation fault. 

Here is my gdb traceback. 
[image]

my cpu details,
[image]

kartikmohta, 2019-05-31:
You can try removing the line:
https://github.com/laboshinl/loam_velodyne/blob/25db5dd5b2c135e779a50a11af0a53434598df7e/CMakeLists.txt#L35
from the `CMakeLists.txt` and then recompile and launch.

AravindChandradoss, 2019-06-01:
Thanks @kartikmohta
It is working now. What exactly has happened?

wwtinwhu, 2019-07-10:
Thanks  @kartikmohta

whitelok, 2019-08-29:
I met this error when I run my code in docker.
As far as I known, g++/gcc compiler is just a translator of C lang to asm. "-march=native" will optimize you code with related CPU instructions(like AVX ). g++/gcc is not responded whether you binary(after compiled with this optimization) can be ran.
so if you want to use this optimization, you can `gcc -c -Q -march=native --help=target | grep march` to get the right one.

whitelok, 2019-08-29:
FYI: https://link.zhihu.com/?target=https%3A//wiki.gentoo.org/wiki/GCC_optimization%23-march

Garrus007, 2020-02-11:
Thank you @kartikmohta! Your advice helped for me and I think it's the best and most simple solution. I tried to uninstall libpcl*, but all ROS packages was removed because PCL is a dependency.

AeroWRX, 2020-06-14:
> I solved it by installing pcl-1.9 from source, and everything is okay since then
>
> You can download the latest release from here: https://github.com/PointCloudLibrary/pcl/releases
>
> And the [...]

Should I install PCL 1.9 specifically to fix this issue or does the latest PCL work also?

miroslavradojevic, 2020-07-19:
Thanks @kartikmohta your solution was helpful.

srinivasrama, 2022-04-12:
Thanks @kartikmohta