laboshinl/loam_velodyne #72: Where is IMU aligned with LIDAR by orientation transformation ?

BrunoEduardoCSantos, 2018-06-24:
Hi everyone, 
I am quite a newbie in this field, so my question could have a simple answer. So, my question where is transformation from IMU to LIDAR reference system?
Thanks,
Bruno

zcg3648806, 2020-01-03:
I have the same problem. Have you solved it？

closed 2020-01-15 by brunoeducsantos

jingwenyi, 2021-08-11:
I did a test. The Z axis of my IMU is vertical and upward is positive. I directly test the source code, and the positioning soon drifted. I found that the gravitational acceleration in the source code is 9.81, indicating that the direction of IMU is vertical and downward is positive. I changed 9.81 to -9.81. Just fine. x. Y axis you can first determine the X and Y axes of lidar in rviz, and then align them with the X and Y axes of IMU.

I modify the code as follows

acc.x() = float(imuIn->linear_ acceleration.y + sin(roll) * cos(pitch) * 9.81);

acc.y() = float(imuIn->linear_ acceleration.z + cos(roll) * cos(pitch) * 9.81);

acc.z() = float(imuIn->linear_ acceleration.x - sin(pitch) * 9.81);