laboshinl/loam_velodyne #82: IMU orientation

andre-nguyen, 2018-08-22:
@laboshinl @facontidavide @StefanGlaser 

Hi guys, tagging you since you seem to know what's going on. 

In the current state of the code, what is the expected orientation of the IMU?

On the original `NSH indoor outdoor` bag file everything works fine. However, for the [gates oscillating motion](http://www.frc.ri.cmu.edu/~jizhang03/Videos/gates_oscillating_motion.mp4) bag that also contains IMU data I do not get good performance.

As the original author noted:
> The input messages from the Velodyne and IMU follow the convention of x- pointing to the front, y- pointing to the left, and z- pointing upward. The IMU messages should be projected to align with the Velodyne frame before sending in.

Does that bag include the projection? Has the projection been included in this code?

Thanks!

andre-nguyen, 2018-08-23:
nvm got it

closed 2018-08-23 by andre-nguyen

anushridixit, 2018-10-02:
Has it been included? What did you find out?

andre-nguyen, 2018-10-02:
The code is just very confusing because its all hard coded to bring things into a camera frame. So just align your axes together and everything will work out. Except the map and pose estimates will have x pointing to upwards Z.

anushridixit, 2018-10-02:
I'm not sure how to do that. Is this post-processing? Or while recording the bag file? Thanks a lot!

andre-nguyen, 2018-10-02:
which ever works for you. You can align your axes by putting your lidar with x forward and your imu with x forward, or you can just rotate it right before feeding it into loam and also rotate the output.

improve100, 2019-01-08:
If I put the imu level, can I use the static tf rotation to get the code running? I tried it and put the i imu accelerate x in the direction of z, the map is still very bad.