laboshinl/loam_velodyne #83: /LaserMapping.cpp:149:28: error: ‘downSizeFilterMap’ was not declared in this scope

elahea2020, 2018-08-22:
Hi, 
When I try to make the file I get this error. 
 In member function ‘virtual bool loam::LaserMapping::setup(ros::NodeHandle&, ros::NodeHandle&)’:
_**catkin_ws/src/loam_velodyne/src/lib/LaserMapping.cpp:149:28: error: ‘downSizeFilterMap’ was not declared in this scope downSizeFilterMap().setLeafSize(fParam, fParam, fParam);**_

Which was not the case before the PR. I looked at the header and the parent class and seems like this function has never been defined.

Would appreciate it if someone could fix this bug.

Thanks

adishavit, 2018-08-23:
Please put a link to the offending line in the commit you are using. Also what build tools are you using?

claydergc, 2018-08-23:
Hi,

I have the same problem. Here is the link to the line: [LaserMapping.cpp:149:28](https://github.com/laboshinl/loam_velodyne/commit/98229e46301d1bc0775ee51a5a11ec35263fa2eb#diff-34c7c0440be78d198bea6a22de06e37aR149)

Thanks

nature1949, 2018-08-23:
I have the same problem and this is a new error that has not been seen before.

TifferPelode, 2018-08-23:
I got the same question, after compiled I got 
`/home/tiffer/test_catkin/src/loam_velodyne-master/src/lib/LaserMapping.cpp:149:28: error: ‘downSizeFilterMap’ was not declared in this scope
          downSizeFilterMap().setLeafSize(fParam, fParam, fParam);
                            ^
[8/18] Building CXX object src/lib/CMa...Files/loam.dir/BasicLaserMapping.cpp.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'loam_velodyne': 
  Command '['/home/tiffer/test_catkin/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1
`
and for `catkin_make` also got false just like elahea2020.` downSizeFilterMap() Not Define`

adishavit, 2018-08-23:
I'll look into it asap.

adishavit, 2018-08-23:
You just need to remove that code. 
The parameter isn't used anywhere in the code.
I made a new PR #84 with the fix.