laboshinl/loam_velodyne #85: Run with a new sensor

TifferPelode, 2018-08-23:
Hi,
    I'm a freshman of 3D SLAM, now I only have a RS-LiDAR-16.
    It publish `pointcloud2` message in ROS.
    
I wonder which part of code can modify to satisfied this sensor?

Tbyo, 2018-09-18:
Hi, friends,
I also have the same question when I use a Hesai-Pandar40P Lidar,
I modified the N-SCAN, include the point_types.h and rawdata.h in Hesai-Ros driver.
However, the error is as follows: 

(rviz:14946): Gtk-WARNING **: Unable to locate theme engine in module_path: "hcengine",
[multiScanRegistration-2] process has died [pid 14928, exit code -11, cmd /home/user/catkin_v/devel/lib/loam_velodyne/multiScanRegistration /multi_scan_points:=/velodyne_points __name:=multiScanRegistration __log:=/home/user/.ros/log/664272ca-ba43-11e8-b0c0-ac1f6b8469dd/multiScanRegistration-2.log].
log file: /home/user/.ros/log/664272ca-ba43-11e8-b0c0-ac1f6b8469dd/multiScanRegistration-2*.log

Does anyone know how to run with a new sensor? 
Thanks

jianrui1, 2018-12-11:
@Tbyo Have you saved this problem? I also want to use Hesai-Pandar40P Lidar.Can you share some erperience to me. Thank you very much.