laboshinl/loam_velodyne #86: Extracting transformation between frames

elahea2020, 2018-08-23:
Hi, 

I am trying to extract the transformation between frames and then fuse all of the frames with another script after loam. After reading the code I figured the transformation must be stored in _aftMappedTrans (https://github.com/laboshinl/loam_velodyne/blob/a06980a3e1619ddc8d34b188ac359a1ad8722a1d/src/lib/LaserMapping.cpp#L305) However when I fuse the frames the transformation seems to be completely wrong. 

Would appreciate it if you could tell me which variable stores the transformation between the frames or transformation of frames to the global world coordinate. 

Thanks

elahea2020, 2018-08-23:
RESOLVED

Shantnu12345, 2018-11-06:
Hi, I am working on the same problem: Extracting transformation between frames. Could you please verify how did you accomplished this?

jianrui1, 2018-12-16:
@elahea2020 @Shantnu12345  Hi, I am doing the same work:extracting transformation between frames.
Have you all succeed? Can you share some experience to me ?
Thanks for advance