laboshinl/loam_velodyne #87: Invoking "make -j4 -l4" failed

zr2539168, 2018-08-24:
[ 59%] Built target cloud_nodelet
[ 65%] Built target cloud_node
[ 70%] Built target transform_node
[ 76%] Built target transform_nodelet
/home/zr/catkin_ws/src/loam_velodyne/src/lib/LaserMapping.cpp: In member function ‘virtual bool loam::LaserMapping::setup(ros::NodeHandle&, ros::NodeHandle&)’:
/home/zr/catkin_ws/src/loam_velodyne/src/lib/LaserMapping.cpp:149:28: error: ‘downSizeFilterMap’ was not declared in this scope
          downSizeFilterMap().setLeafSize(fParam, fParam, fParam);
                            ^
loam_velodyne/src/lib/CMakeFiles/loam.dir/build.make:182: recipe for target 'loam_velodyne/src/lib/CMakeFiles/loam.dir/LaserMapping.cpp.o' failed
make[2]: *** [loam_velodyne/src/lib/CMakeFiles/loam.dir/LaserMapping.cpp.o] Error 1
CMakeFiles/Makefile2:3080: recipe for target 'loam_velodyne/src/lib/CMakeFiles/loam.dir/all' failed
make[1]: *** [loam_velodyne/src/lib/CMakeFiles/loam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Anybody knows what is the problem? Many thanks

zr2539168, 2018-08-24:
it happens when run
catkin_make -DCMAKE_BUILD_TYPE=Release

adishavit, 2018-08-24:
See or #84.
https://github.com/laboshinl/loam_velodyne/pull/84