laboshinl/loam_velodyne #90: error: unable to find numeric literal operator ‘operator""Q’

strualo, 2018-09-01:
When i build,it has a error: unable to find numeric literal operator ‘operator""Q’.What can i do?

la9881275, 2019-01-07:
I come across the same problem.
/usr/include/boost/math/constants/constants.hpp:316:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(zeta_two, 1.644934066848226436472415166646025189e+00, "1.64493406684822643647241516664602518921894990120679843773555822937000747040320087383362890061975870530400431896e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:316:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_zeta_three<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:317:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(zeta_three, 1.202056903159594285399738161511449990e+00, "1.20205690315959428539973816151144999076498629234049888179227155534183820578631309018645587360933525814619915780e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:317:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_catalan<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:318:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(catalan, 9.159655941772190150546035149323841107e-01, "9.15965594177219015054603514932384110774149374281672134266498119621763019776254769479356512926115106248574422619e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:318:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_glaisher<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:319:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(glaisher, 1.282427129100622636875342568869791727e+00, "1.28242712910062263687534256886979172776768892732500119206374002174040630885882646112973649195820237439420646120e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:319:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_khinchin<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:320:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(khinchin, 2.685452001065306445309714835481795693e+00, "2.68545200106530644530971483548179569382038229399446295305115234555721885953715200280114117493184769799515346591e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:320:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_extreme_value_skewness<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:321:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(extreme_value_skewness, 1.139547099404648657492793019389846112e+00, "1.13954709940464865749279301938984611208759979583655182472165571008524800770607068570718754688693851501894272049e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:321:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_rayleigh_skewness<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:322:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(rayleigh_skewness, 6.311106578189371381918993515442277798e-01, "6.31110657818937138191899351544227779844042203134719497658094585692926819617473725459905027032537306794400047264e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:322:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_rayleigh_kurtosis<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:323:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(rayleigh_kurtosis, 3.245089300687638062848660410619754415e+00, "3.24508930068763806284866041061975441541706673178920936177133764493367904540874159051490619368679348977426462633e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:323:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_rayleigh_kurtosis_excess<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:324:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(rayleigh_kurtosis_excess, 2.450893006876380628486604106197544154e-01, "2.45089300687638062848660410619754415417066731789209361771337644933679045408741590514906193686793489774264626328e-01")

GSRLW, 2020-05-21:
I found it ,too. but I have no idea to solve it.

GSRLW, 2020-05-21:
I think this will work by change your Flags,  I changed my flags from -std=c+=11  to   -std=gnu++11 .