laboshinl/loam_velodyne #92: not working with my own data

zr2539168, 2018-09-28:
I use a VLP16 lidar to run loam (without IMU or GPS), it seems not working, the graph on rviz rotates chaostically. I tried bag file, pcap file and sensor stream, none works appropriately. Is IMU module required or other issues? Many thanks.

zr2539168, 2018-09-28:
btw the samples work well

anushridixit, 2018-10-02:
@zr2539168 were you able to solve this? I am having the same issue

zr2539168, 2018-10-24:
@anushridixit Sorry, not solved..

anushridixit, 2018-10-24:
I solved this problem. For me, the Velodyne motor rotation rate had to be manually set to 600 RPM on the sensor's network address ([section 1.3 here](http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16)). It sometimes automatically gets set to 300 RPM even if you have changed it during the initial setup. 

The samples work because they are recorded with motor RPM at 600.
Hope this helps!