laboshinl/loam_velodyne #94: Reproducing KITTI benchmark results

rdube, 2018-10-25:
Hi guys, did anyone manage to reproduce the results for the KITTI benchmark? We are looking to achieve this on our end and would be happy to get in touch with anyone who worked towards this. Thanks!

Shantnu12345, 2018-11-07:
I am working towards this. Please ping me on shantnukakkar1992@gmail.com

Xiangzhaohong, 2018-12-01:
I am working on this. What should I do to reproduce the results?
Thanks!!

jianrui1, 2018-12-11:
@Shantnu12345  Are you reproduce the result? Can you help me out 
Thank you

andre-nguyen, 2019-02-17:
@rdube Read "Low-drift and Real-time Lidar Odometry and Mapping", the implementation submitted to Kitti differs in a few ways to this implementation.

PeiyueYu, 2021-03-19:
> @rdube Read "Low-drift and Real-time Lidar Odometry and Mapping", the implementation submitted to Kitti differs in a few ways to this implementation.

I have read ther paper, but I didn't find the implementation submitted to Kitti in this paper, could you please point out the section?