laboshinl/loam_velodyne #97: Scan by scan odometry

Shantnu12345, 2018-11-07:
Fellow researchers,

Could anyone please explain how to use LOAM to get odometry between the frames in the input pointclouds. There are 12000 frames in my input pointclouds. So, correspondingly, I would like 11999 odometry messages, where ith message is the odometry of ith frame in the first frame's coordinate system. 

Currently, I am getting around 14000 messages for /integrated_to_init. I would like to have 11999 messages instead representing the odometry of each frame.

Thank you for your help.

suchet4620, 2020-02-21:
> Fellow researchers,
> 
> Could anyone please explain how to use LOAM to get odometry between the frames. There are 12000 frames in my input pointcloud pcap file (which I am publishing using velodyne_pointcloud node). So, correspondingly, I would like 11999 odometry messages, where ith message is the odometry of ith frame in the first frame's coordinate system.
> 
> Currently, I am getting around 14000 messages for /integrated_to_init. I would like to have 11999 messages instead representing the odometry of each frame.
> 
> Your help is really appreciated.
> 
> Thanks,
> Shantnu

Hi, Shantnu
Would you mind to explain in details how can you get odometry message (for integrated_to_init) between the frames?

Thanks,
Suchet

Shantnu12345, 2020-02-21:
Hey Suchet,

First, your machine might be skipping frames which is causing the number of
messages in integrated_to_init to be so low. I faced this problem when I
was running LOAM on a VM, but it went away when I ran it on native Ubuntu.

Secondly, there is a condition in LOAM that stops prevents of first 20-25
integrated_to_init.  That condition will lead to publishing of
approximately 11976 (11999-23) messages. You need to comment out those
conditions to get all 11999 integrated_to_init messages. That condition is
in src/lib/MultiScanRegistration.cpp:
  if (_systemDelay > 0)
  {
    --_systemDelay;
    return;
  }

Best,
Shantnu

suchet4620, 2020-02-23:
> Hey Suchet, First, your machine might be skipping frames which is causing the number of messages in integrated_to_init to be so low. I faced this problem when I was running LOAM on a VM, but it went [...]

Hi, Shantnu

Thank you for your quick reply, basically I am a beginner for this field, after several weeks I can run LOAM on native Ubuntu 18.04, everything looking great as it would be (e.g. I can get velodyne_points). But in order to extract/get an odometry messages in integrated_to_init, I have nothing saved in my created folder, even I try to comment out those lines in src/lib/MultiScanRegistration.cpp as you mentioned. 

And I got some error message as below after commanding by 'rosrun pcl_ros pointcloud_to_pcd input:=/integrated_to_init' :

[ERROR] [1582427493.983276695]: Client [/pointcloud_to_pcd_1582427493768252954] wants topic /integrated_to_init to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7]. Dropping connection. 

Would you please suggest me, Is there any wrong with my command or not?, or did I miss some points?

Thank you,
Suchet