laboshinl/loam_velodyne #98: How could I compare the pose result of LOAM with the ground truth?

AbnerCSZ, 2018-11-10:
How could I compare the pose result of mine algorithm(SLAM/Odometry) with the ground truth?
Just like tum rgb-d tool:
[Useful tools for the RGB-D benchmark](https://vision.in.tum.de/data/datasets/rgbd-dataset/tools)  

In other words, how could I calculate the ate(ABSOLUTE TRAJECTORY ERROR) and RELATIVE POSE ERROR(rpe) between My pose result from LOAM and the benchmark?

Thanks!

closed 2018-12-07 by AbnerCSZ

HongweiSunny, 2018-12-23:
Hi,is your problem solved?
I want to compare the ground-truth and poses from loam too.

langelesl, 2019-03-11:
I don't know if you have solved the problem. When I ran "$ rostopic echo /lidar_odom_to_init/pose/pose > result.txt" in a new terminal, I could get a txt file that contains position (x,y,z) and orientation (x,y,z,w)information of each frame, just like the result from ORB_SLAM2. Then I could evaluate the result with the ground truth file of KITTI. But seems like the result I got from LOAM is less accurate than the trajectory from ORB. Do you guys have a better idea for method of evaluation or tuning for a better result like Zhang did on KITTI Odom Benchmark?

claydergc, 2019-03-11:
@langelesl the code of LOAM is designed for raw data. The KITTI dataset has 2 versions: a raw one and a distortion-free dataset. Which one are you using? If you are trying the distortion-free dataset, you can check my repo https://github.com/claydergc/loam_velodyne_kitti_ros from which I got good results. About tunning, there are parameters like the minimum number of features extracted to start the LM optimization that can be changed, given that the KITTI data uses an HDL-64 and this version is for the VLP-16.

langelesl, 2019-03-11:
Thank you very much for your explanation and suggestion. @claydergc I am using distortion-free dataset. I will try your repo as soon as possible.

LongruiDong, 2019-07-19:
> Thank you very much for your explanation and suggestion. @claydergc I am using distortion-free dataset. I will try your repo as soon as possible.

Hi~
I also want to reproduce LOAM result on kitti 00-10 Sequences, does these training sequences are "distortion-free dataset" as you said ?
Thanks~

claydergc, 2019-07-19:
Yes, they are distorion-free datataset.

xuwuzhou, 2019-12-14:
> Yes, they are distorion-free datataset.

Hi~
I have tried your repo in 00 Sequence
And I set the frequence at 1hz;
Although I get a not bad result ,I find that the poses out is the ground truth poses turned right 270degrees,that's for why?