laboshinl/loam_velodyne #100: Localization in existing map

Jackey-Huo, 2018-11-25:
@laboshinl Thank you so much for the amazing work! I noticed that there seem to be no capability to localize on an existing map.

Is there any schedule to develop relocalization? Or is a pr for the feature welcomed :)?

NTUwilliam, 2018-12-24:
Hi all 

I have the same question with @Jackey-Huo. Can `relocalization` (localize in an existing point cloud map) function be added to the future version of **loam**? 
If this can be realized, **loam** can be much more powerful :)

Anyway, thanks for your great work!
@laboshinl @StefanGlaser @facontidavide @evenator

NTUwilliam, 2018-12-27:
Hi all

I have the same question with @Jackey-Huo. Can relocalization (localize in an existing point cloud map) function be added to the future version of loam?
If this can be realized, loam can be much more powerful :)

Anyway, thanks for your great work!
@laboshinl @StefanGlaser @facontidavide @evenator