laboshinl/loam_velodyne #101: Link error: undefined reference of ros::init(...), ros::NodeHandle(...) ....

wykxwyc, 2018-12-01:
I build the project, but there are many Link Errors that I can not fix. Can anyone give provide some advice?
Thanks very much!

### **And here is my build progress:**
**note: some Chinese translated into English:
"在函数‘main’中": in the main function
"未定义的引用": undefined reference
"错误": error
"返回 ": return
"正在等待未完成的任务....": waiting for the tasks to be finished...**

robot@robot:~/loam_ws$ catkin_make -DCMAKE_BUILD_TYPE=Release 
Base path: /home/robot/loam_ws
Source space: /home/robot/loam_ws/src
Build space: /home/robot/loam_ws/build
Devel space: /home/robot/loam_ws/devel
Install space: /home/robot/loam_ws/install
####
#### Running command: "cmake /home/robot/loam_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/robot/loam_ws/devel -DCMAKE_INSTALL_PREFIX=/home/robot/loam_ws/install -G Unix Makefiles" in "/home/robot/loam_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/robot/loam_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/robot/catkin/devel;/opt/ros/indigo
-- This workspace overlays: /home/robot/catkin/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/robot/loam_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
WARNING: Package name "LC" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - loam_velodyne
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'loam_velodyne'
-- ==> add_subdirectory(loam_velodyne)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- Checking for module 'openni-dev'
--   No package 'openni-dev' found
-- Checking for module 'openni-dev'
--   No package 'openni-dev' found
-- Checking for module 'openni-dev'
--   No package 'openni-dev' found
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_KEYPOINTS
-- looking for PCL_GEOMETRY
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
[... 59 lines cut ...]
multi_scan_registration_node.cpp:(.text.startup+0x15a)：对‘ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::hasParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::getParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::getParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::getParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)’未定义的引用
collect2: 错误：ld 返回 1
make[2]: *** [/home/robot/loam_ws/devel/lib/loam_velodyne/multiScanRegistration] 错误 1
make[1]: *** [loam_velodyne/CMakeFiles/multiScanRegistration.dir/all] 错误 2
[100%] Linking CXX executable /home/robot/loam_ws/devel/lib/loam_velodyne/laserMapping
CMakeFiles/laserMapping.dir/src/laser_mapping_node.cpp.o：在函数‘main’中：
laser_mapping_node.cpp:(.text.startup+0x34)：对‘ros::init(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)’未定义的引用
laser_mapping_node.cpp:(.text.startup+0xc7)：对‘ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)’未定义的引用
laser_mapping_node.cpp:(.text.startup+0x15a)：对‘ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::hasParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::getParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::getParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::NodeHandle::getParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const’未定义的引用
/home/robot/loam_ws/devel/lib/libloam.so：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)’未定义的引用
collect2: 错误：ld 返回 1
make[2]: *** [/home/robot/loam_ws/devel/lib/loam_velodyne/laserMapping] 错误 1
make[1]: *** [loam_velodyne/CMakeFiles/laserMapping.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j4 -l4" failed
robot@robot:~/loam_ws$

asmand001, 2019-04-18:
有人解决啦吗？

wykxwyc, 2019-04-18:
> 有人解决啦吗？

把编译器改回gcc4.8（不知道你是不是和我一样将编译器改成了高版本）

asmand001, 2019-04-18:
谢谢啊，是的，我的4.9    后来还试了5 10
不知道你改成4.8解决了吗？

wykxwyc, 2019-04-18:
是的，解决了。

smmmmi, 2020-02-04:
sudo apt-get install gcc-5 g++-5
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 10
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-5 10
sudo update-alternatives --config gcc
choose selection number