laboshinl/loam_velodyne #126: Error occurs when play the bag of VLP16

LiaoJunchan, 2019-03-20:
Successfully run the launch file (loam_velodyne.launch), but once I play the bag of VLP16, error below occurs. What does this error means and how to solve it? 
[multiScanRegistration-2] process has died [pid 24636, exit code -11, cmd /home/fergus/loam_velodyne/devel/lib/loam_velodyne/multiScanRegistration /multi_scan_points:=/velodyne_points __name:=multiScanRegistration __log:=/home/fergus/.ros/log/6c595248-4ae7-11e9-a43a-b46d83213513/multiScanRegistration-2.log].
log file: /home/fergus/.ros/log/6c595248-4ae7-11e9-a43a-b46d83213513/multiScanRegistration-2*.log

Passionzq, 2019-04-25:
Hello, the download page of BAG files is missing now. Can you send me the BAG files you have download? My e-mail address is 546035409@qq.com     Thank you!

he-guo, 2019-10-28:
hello,i have encountered the same problem.Have you solved it?
Thanks.

DavidHan008, 2020-03-19:
我也遇到了同样的问题,请问怎么解决

DavidHan008, 2020-03-20:
作者的repro的readme里面已经有介绍了,必须源码编译的pcl库才行

jingshaojing, 2020-07-12:
> 作者的repro的readme里面已经有介绍了,必须源码编译的请问已经源码编译过了，仍旧遇到这个问题怎么解决

jingshaojing, 2020-07-12:
> 作者的repro的readme里面已经有介绍了,必须源码编译的pcl库才行

已源码编译了pcl库，仍旧报错，请问您知道什么原因么

miroslavradojevic, 2020-07-18:
I am experiencing the same issue, tried several .bag files including nsh_indoor_outdoor.bag. 
In BasicScanRegistration::extractFeatures() method of BasicScanRegistration.cpp 
there is a for loop for (size_t i = beginIdx; i < nScans; i++) {...} which executes only once (i=0), finishes the whole block and interrupts the exectuion of the multiScanRegistration node after completing loop exactly once. I could not notice any errors in code, the for-loop simply stops before entering 2nd loop.

miroslavradojevic, 2020-07-19:
The problem turned out to be related to the issue [#71](https://github.com/laboshinl/loam_velodyne/issues/71). pcl lib incompatbility. 
Although they suggest removing pcl lib and building it from source, I followed advice to remove line
add_definitions( -march=native ) 
from loam_velodyne/CMakeLists.txt
as suggested in [issue #71 thread](https://github.com/laboshinl/loam_velodyne/issues/71#issuecomment-497567752) 
Just comment the line, compile and the node will not break with execution.

jingshaojing, 2020-07-23:
> The problem turned out to be related to the issue [#71](https://github.com/laboshinl/loam_velodyne/issues/71). pcl lib incompatbility.
> Although they suggest removing pcl lib and building it from [...]

thanks.Though i do not know what happened, when i posted the question ,I happened to succes,but todady, i changed to other computer and do like last time,but fail. I do it as your follows, and success