laboshinl/loam_velodyne #128: Exporting map

space03, 2019-03-29:
Hi. 

I want to export the final map to pcd files. 
From issues, i found the following comment.
https://github.com/laboshinl/loam_velodyne/issues/3#issuecomment-223254071

1st Terminal: _roslaunch loam_velodyne loam_velodyne.launch_ 
2nd Terminal: _rosbag play ~/Downloads/MY_BAG_FILE_NAME.bag_ 
3rd Terminal:  _rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered_

but I'm in trouble. In 3rd terminal, exporting pcd file command is not working. 

rosrun pcl_ros pcd_to_pointcloud input:=/velodyne_cloud_registered
[ INFO] [1553843254.665227026]: Publishing data on topic /cloud_pcd with frame_id /base_link.
Segmentation fault (core dumped)

Please help me. Thank you.

space03, 2019-04-01:
It is solved by rebooting the computer.

But i have other problems. I want to save the whole(full size) map.
But " rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered" is saved only current window point cloud information. 
Maybe it is similar problem with https://github.com/laboshinl/loam_velodyne/issues/41#issuecomment-363551983.
@StefanGlaser , @Shantnu12345
I dont have any idea of how to change original code of loam, laserMapping.cpp(~/catkin_ws/src/loam_velodyne/src/lib).
please help me.

mhaboali, 2020-08-17:
> It is solved by rebooting the computer.
> 
> But i have other problems. I want to save the whole(full size) map.
> But " rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered" is saved only current window point cloud information.
> Maybe it is similar problem with [#41 (comment)](https://github.com/laboshinl/loam_velodyne/issues/41#issuecomment-363551983).
> @StefanGlaser , @Shantnu12345
> I dont have any idea of how to change original code of loam, laserMapping.cpp(~/catkin_ws/src/loam_velodyne/src/lib).
> And is it correct that exporting map to pcd command is " rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered"?
> please help me.

I have the same issue, it captures only the recent points in 0.pcd file