laboshinl/loam_velodyne #129: The mapping result become sparse

tuduku, 2019-04-01:
I have run the code and get a good result before
But it seems that some wrong happen now.

The odometry jump to other plane sometimes
[image]

And the Mapping result become sparse 
[image]

I cloned the new code and run according to README ,but it dose not work.
I didn't find any wrong in the code.My computer is  Intel Core i7-8th Gen CPU ,GTX1060 and I ran the code in Ubuntu 16.04 . I can get the true mapping result just several days ago.
Please help me or give me a hint if it won't take you long,thanks.

Bzhnja, 2019-04-03:
Have you compiled the ros workspace with`catkin_make -DCMAKE_BUILD_TYPE=Release` ? 
The debug version and release version have distinct performance.

tuduku, 2019-04-03:
> Have you compiled the ros workspace with`catkin_make -DCMAKE_BUILD_TYPE=Release` ?
> The debug version and release version have distinct performance.

thanks a lot!  it works. and sorry for my careless.

geoeo, 2019-08-08:
@tuduku What dataset are you using?