laboshinl/loam_velodyne #134: Odometrie orientation

lchop, 2019-04-24:
Hello,

I have an issue with the odometrie orientation.
https://www.youtube.com/watch?v=YTCnmP5RBuE
The yellow arrow is not pointing in the direction where the robot is going but instead it is "horizontal". I have the same issue here:
[image]
I have correct it applying a PI/2 rotation to the odom message:
[image]

But I am not happy with this fixe. Anyone has experienced this issue ? Is this even an issue ?

YoshuaNava, 2019-04-24:
This is due to LOAM using a camera coordinate frame convention. It's described in the author's papers.

If you want to have a quick fix, just rotate the output poses/clouds of LOAM to your desired coord. frame orientation.

For a more elaborate, long term fix you'd need to dive deeper into the code. It performs all the calculations in the camera coordinate frame, so there is quite that would need to be changed.

lchop, 2019-04-25:
Hello,

Thank you for your answers. I have read the author's paper, two frames are mentionned, a world frame W and a Laser Frame L. Just to be sure, the camera frame convention is applied to the 3D laser Frame, is that correct ?

YoshuaNava, 2019-04-27:
> Hello,
> 
> Thank you for your answers. I have read the author's paper, two frames are mentionned, a world frame W and a Laser Frame L. Just to be sure, the camera frame convention is applied to the 3D laser Frame, is that correct ?

And to the IMU.

closed 2019-04-29 by lchop