laboshinl/loam_velodyne #135: How to  read from velodyne VLP16 sample pcap?

Passionzq, 2019-04-25:
Hello everyone:
I met some problem at step "Running" and I need your help.
After building with catkin, I found that I cannot download the velodyne data from "VLP16 rosbag"(Because the page does not exist now). So I download three PCAP file from "VLP16 sample pcap".
But when i execute
`roslaunch velodyne_pointcloud VLP16_points.launch pcap:="$HOME/Downloads/velodyne.pcap"
`

It displays
`RLException: [VLP16_points.launch] is neither a launch file in package [velodyne_pointcloud] nor is [velodyne_pointcloud] a launch file name
The traceback for the exception was written to the log file
`

So how can I user PCAP file? 
Should I change them to BAG file at first?

LongruiDong, 2019-07-19:
> Hello everyone:
> I met some problem at step "Running" and I need your help.
> After building with catkin, I found that I cannot download the velodyne data from "VLP16 rosbag"(Because the page does not exist now). So I download three PCAP file from "VLP16 sample pcap".
> But when i execute
> `roslaunch velodyne_pointcloud VLP16_points.launch pcap:="$HOME/Downloads/velodyne.pcap" `
> 
> It displays
> `RLException: [VLP16_points.launch] is neither a launch file in package [velodyne_pointcloud] nor is [velodyne_pointcloud] a launch file name The traceback for the exception was written to the log file `
> 
> So how can I use PCAP file?
> Should I change them to BAG file at first?
> Or could you please send me the BAG file? (e-mail: [546035409@qq.com](mailto:546035409@qq.com))
look this
https://github.com/laboshinl/loam_velodyne/issues/3#issuecomment-215067765

jingshaojing, 2020-07-10:
> > Hello everyone:
> > I met some problem at step "Running" and I need your help.
> > After building with catkin, I found that I cannot download the velodyne data from "VLP16 rosbag"(Because the page does not exist now). So I download three PCAP file from "VLP16 sample pcap".
> > But when i execute
> > `roslaunch velodyne_pointcloud VLP16_points.launch pcap:="$HOME/Downloads/velodyne.pcap" `
> > It displays
> > `RLException: [VLP16_points.launch] is neither a launch file in package [velodyne_pointcloud] nor is [velodyne_pointcloud] a launch file name The traceback for the exception was written to the log file `
> > So how can I use PCAP file?
> > Should I change them to BAG file at first?
> > Or could you please send me the BAG file? (e-mail: [546035409@qq.com](mailto:546035409@qq.com))
> 
> look this
> [#3 (comment)](https://github.com/laboshinl/loam_velodyne/issues/3#issuecomment-215067765)

Have you solved this problem?

jingshaojing, 2020-07-10:
> Hello everyone:
> I met some problem at step "Running" and I need your help.
> After building with catkin, I found that I cannot download the velodyne data from "VLP16 rosbag"(Because the page does not [...]

Have you solved this problem?