laboshinl/loam_velodyne #148: [scanRegistration] died when rosbag play .bag file collected  from leishenc16 lidar.

lwbdegit, 2019-10-04:
Terminal says：
[scanRegistration-2] process has died [pid 17156, exit code -11, cmd /home/imcngx/loam2/devel/lib/loam_velodyne/scanRegistration __name:=scanRegistration __log:=/home/imcngx/.ros/log/85f29d2c-e678-11e9-9e53-2b8176dc7a37/scanRegistration-2.log].
log file: /home/imcngx/.ros/log/85f29d2c-e678-11e9-9e53-2b8176dc7a37/scanRegistration-2*.log

he-guo, 2019-10-28:
hello,i have encountered the same problem.Have you solved it?
Thanks.

lwbdegit, 2019-10-28:
please try to enlarge the array length from 40000 to 80000  ------------------ 原始邮件 ------------------

 
&mdash;

he-guo, 2019-10-28:
Thank you for you reply,I think it is.I started the VLP16 driver file with creating the thread,this problem has occurred,but playing packet will not have this problem.If I want to collect point cloud date in real time,I must open the VLP16 driver node,and Threads will be generated.Do you have any good suggestions for this?
thank you very much.

lwbdegit, 2019-10-28:
Sorry，I am not very clear about the problem you mentioned, so I may not be able to help you solve it for the time being. You can ask someone else.  The problem I encountered at that time was that I could run loam with VLP data set, but the leishenc16 radar or the data set recorded with it would have the problem of scanregistration node crashing. At that time, I suspected that the data volume of LS radar was too large, so I tried to increase the array length.  The author uses vlp16 radar. Your hardware and software are the same as those of the author. How could this problem happen?------------------ 原始邮件 ------------------

he-guo, 2019-10-28:
Yes,using VLP data set,I have not encountered problem.But I'd like to collect data in real time for mapping,and this problem will occur.In other words,I will find this problem when running both loam and VLP lider driver.it's OK when I run loam and rosbag VLP data set.