laboshinl/loam_velodyne #155: Some questions about system performance

zcg3648806, 2020-04-06:
I read some of the papers on slam, and I found that in these papers the performance of slam was usually evaluated with the average relative translation （%） and rotation （deg/m）, It's just like the kitti list shows：
77239946-483b8b00-6c1b-11ea-8c01-f2276333e6ae
How to get this result?
In fact, I don't quite understand the meaning of these two evaluation indexes, so I don't know what the formula is for this parameter.
I have used evo to test the slam performance results, and can get ape and rpe, but these are not the above two evaluation parameters.
I just want to know what their formula is. Is it the absolute error of the final frame divided by the total distance？

RuanJY, 2021-03-01:
Just look at the text behind this table, there are information you need. 
"our evaluation computes translational and rotational errors for all possible subsequences of length (100,...,800) meters. The evaluation table below ranks methods according to the average of those values, where errors are measured in percent (for translation) and in degrees per meter (for rotation). A more detailed comparison for different trajectory lengths and driving speeds can be found in the plots underneath"

lgm051, 2021-12-09:
> I read some of the papers on slam, and I found that in these papers the performance of slam was usually evaluated with the average relative translation （%） and rotation （deg/m）, It's just like the kitti list shows： [image] How to get this result? In fact, I don't quite understand the meaning of these two evaluation indexes, so I don't know what the formula is for this parameter. I have used evo to test the slam performance results, and can get ape and rpe, but these are not the above two evaluation parameters. I just want to know what their formula is. Is it the absolute error of the final frame divided by the total distance？

Hi, brother, do you know how to use Evo to measure these two parameters of Kitti? Can you give me some advises?

RuanJY, 2021-12-10:
> 

Read their README. btw, I use this one: https://github.com/LeoQLi/KITTI_odometry_evaluation_tool, it works ok.

zcg3648806, 2021-12-31:
> Hi, brother, do you know how to use Evo to measure these two parameters of Kitti? Can you give me some advises?

No, I just get rpe,ape,rte,ate with the EVO.