laboshinl/loam_velodyne #157: Error transforming odometry 'Odometry' from frame '/camera_init' to frame 'camera_init'

mberrey, 2020-06-11:
When trying to launch a pcap file or play a bag file, they begin processing on rviz but the error "Error transforming odometry 'Odometry' from frame '/camera_init' to frame 'camera_init'" occurs each tf, causing a skewed tf path and no pointcloud in rviz. Any tips/tricks on how to solve this? I was trying to look at the tf tree and rqt graph to troubleshoot, no luck though. Thanks.

mberrey, 2020-06-11:
Could it be because of the broadcaster rate, seen below:
[frames.pdf](https://github.com/laboshinl/loam_velodyne/files/4762536/frames.pdf)

laurojljunior, 2020-10-06:
Hi @mberrey how did you fix that? Thanks.

MontyTHall, 2020-12-05:
@laurojljunior , yes, I'm in the same boat as you.  @mberrey did you find a solution?

laurojljunior, 2020-12-05:
Hey @MontyTHall. Seems that’s a problem with recent version of RViz. Just change the camera_init frame on rviz from /camera_init to //camera_init. That solved my problem.

MontyTHall, 2020-12-05:
@laurojljunior great, I can see the odometry path.  but I can't seem to see the point cloud.  You get a point cloud?

wrotcat, 2020-12-06:
@MontyTHall  Hi, i got the same problem. Did you find a solution to get the point cloud?

MontyTHall, 2020-12-06:
@wrotcat, no I still cannot see the point cloud.

wrotcat, 2020-12-09:
@MontyTHall  Hey. Maybe you could 'catkin_make' it again, and change /camera_init to //camera_init.  I don't know what happened exactly but the point clouds work now.

JackBorer, 2020-12-13:
I was having a problem visualizing the data in RVIZ in a branch my lab uses [LeGO-LOAM](https://github.com/ut-amrl/LeGO-LOAM/tree/jack-devel). On the jack-devel branch you can see how I removed all of the forward slashes from the hard coded tf frames in a commit I made today. Rebuild after making those changes and it worked like a charm.

This may not be the best way to solve the problem, but as I understand it the ROS tf tree is moving away from the forward slash. I hope someone here could provide some more clarity on that or give a better solution to this problem then doing "//camera_init".

yshry, 2020-12-14:
@MontyTHall I was in the same situation and by running 'catkin_make -DCMAKE_BUILD_TYPE=Release' solved my case. I forgot the option last time.

yqmy0814, 2021-05-14:
I tried to change "/camera_init" to "//camera_init" but it doesn't work, with terminal warning "Invalid argument "/camera_init" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like:"
However changing "/camera_init" to "camera_init" can solve this problem.

clobot-noah, 2021-05-25:
I tried to change "/camera_init" to "camera_init" at the code. I searched it and change it into all codes.

darshank528, 2021-05-29:
> I tried to change "/camera_init" to "//camera_init" but it doesn't work, with terminal warning "Invalid argument "/camera_init" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like:"
> However changing "/camera_init" to "camera_init" can solve this problem.

Hi yqmy0814,
Can you please explain where do I have to go for making this change, in Rviz window only or some config file?

clobot-noah, 2021-05-31:
I'll try it.
case 1) Change [Global Options] - [Fixed Frame] at the Displays tap of Rviz.
Change "camera_init" to "//camera_init". but, when I tried case 1), it is always executed.
case 2) Change the code.
You search "/  into the loam code. you should change the code.(LaserMapping.cpp, LaserOdomertyr.cpp, TransformMaintenance.cpp)
ex) LaserMapping.cpp - LaserMapping::LaserMapping(const float& scanPeriod, const size_t& maxIterations)           
  odomAftMapped.header.frame_id = "/camera_init";  --> odomAftMapped.header.frame_id = "camera_init";

clobot-noah, 2021-05-31:
> > I tried to change "/camera_init" to "//camera_init" but it doesn't work, with terminal warning "Invalid argument "/camera_init" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like:"
> > However changing "/camera_init" to "camera_init" can solve this problem.
> 
> Hi yqmy0814,
> Can you please explain where do I have to go for making this change, in Rviz window only or some config file?
> I am getting the same error "Invalid argument "/camera_init" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like:". Please help me out here.

You will try it. 
case 1) Change [Global Options] - [Fixed Frame] at the Displays tap of Rviz.
Change "camera_init" to "//camera_init". but, when I tried case 1), it is always executed.
case 2) Change the code.
You search "/  into the loam code. you should change the code.(LaserMapping.cpp, LaserOdomertyr.cpp, TransformMaintenance.cpp)
ex) LaserMapping.cpp - LaserMapping::LaserMapping(const float& scanPeriod, const size_t& maxIterations)           
  odomAftMapped.header.frame_id = "/camera_init";  --> odomAftMapped.header.frame_id = "camera_init";

ThakurSarveshGit, 2024-01-28:
> Hey @MontyTHall. Seems that’s a problem with recent version of RViz. Just change the camera_init frame on rviz from /camera_init to //camera_init. That solved my problem.

How to save cameras pose?