laboshinl/loam_velodyne #158: Segmentation Fault on "multiScanRegistration" node

Marcus-Davi, 2020-07-28:
Using the "nsh_indoor_outdoor.bag" bagfile provided by the author, I've launched "loam_velodyne.launch". After a few seconds, the node "multiScanRegistration" crashes with a segmentation fault. I haven't changed any parameters for this test. Has anyone experienced the same ?  I've checked the node topic and remaps, and it crashes as soon as the /velodyne_poins topic is published by the bagfile.

[image]

Doublehes, 2020-09-10:
I have met the same problem with you, have you solved it?

Marcus-Davi, 2020-09-10:
not yet, I've dropped the tests ! but I'll be resuming soon

cynosure4sure, 2020-09-28:
Is this issue related?
https://github.com/laboshinl/loam_velodyne/pull/139

Doublehes, 2020-11-18:
maybe something wrong with my pcap file, because my pcap file was not captured by veloview.

chukhoi61, 2024-01-16:
> Is this issue related? #139

I did as you said but still got the error