laboshinl/loam_velodyne #159: Questions about the interpolation of laser point.

cpkingw, 2020-09-28:
It seems that **_scanTime** is the timestamp of the previous laser frame, because the update of the **_scanTime** or **_sweepStart** in the **reset()** is behind the the interpolation of laser point via **projectPointToStartOfSweep()**. When doing interpolation of point in current laser frame, Shouldn't it be the timestamp of the current frame? 
```
void BasicScanRegistration::interpolateIMUStateFor(const float &relTime, IMUState &outputState)
{
  double timeDiff = toSec(_scanTime - _imuHistory[_imuIdx].stamp) + relTime;
  while (_imuIdx < _imuHistory.size() - 1 && timeDiff > 0) {
    _imuIdx++;
    timeDiff = toSec(_scanTime - _imuHistory[_imuIdx].stamp) + relTime;
  }
  ...
}
```

```
void BasicScanRegistration::reset(const Time& scanTime)
{
  _scanTime = scanTime;

  // re-initialize IMU start index and state
  _imuIdx = 0;
  if (hasIMUData()) {
    interpolateIMUStateFor(0, _imuStart);
  }

  // clear internal cloud buffers at the beginning of a sweep
  if (true/*newSweep*/) {
    _sweepStart = scanTime;
    ...
  }
}
```

arshhhly, 2021-01-08:
I think it is a mistake.

gackes, 2021-07-26:
Have you solved the problem? I have the same doubt. @cpkingw

cpkingw, 2021-08-03:
> Have you solved the problem? I have the same doubt. @cpkingw

I think you can try to update the **_scanTime** in **projectPointToStartOfSweep()** firstly by adding **_scanTime** as a new argument like **reset()**