laboshinl/loam_velodyne #160: VLP16 rosbag not longer avaliable

Cristian-wp, 2020-11-18:
Hello,
I am starting to work with this package but I can not get the file velodyne.bag because when I click on the following  [link](https://frc.ri.cmu.edu/~jizhang03/Datasets/) to download it, an error occour:  "Sorry, but the page you are trying to view does not exist." 

[image]

Probably the file was removed, can someone provide me this bag file to test the package?

Thanks in advance to everyone!
Cristian

Marcus-Davi, 2021-04-24:
Same here, please can someone publish the bagfile?

sterngerlach, 2021-08-13:
I recently downloaded bag files from here (`schenley_bridge.bag` was not avilable though):
https://web.archive.org/web/20161031231526/https://frc.ri.cmu.edu/~jizhang03/Datasets/

harisbinyousaf24, 2023-06-06:
The link you provided https://web.archive.org/web/20161031231526/https://frc.ri.cmu.edu/~jizhang03/Datasets/
Doesn't contain tf data for the rosbag and i am encountring the issue in this regard. Can you provide the loam package that you used for the rosbag or modified launch file if you tweaked for that purpose? Woud highly appreciate your assistance in this regard