laboshinl/loam_velodyne #161: comput the curvature

DarrenWong, 2021-02-16:
Dear All,

A question related to feature extraction.  The implementation is here,
https://github.com/laboshinl/loam_velodyne/blob/25db5dd5b2c135e779a50a11af0a53434598df7e/src/lib/BasicScanRegistration.cpp#L306
But in the origin LOAM, the Eq(1) should be normalized w.r.t the distance to the lidar center to eliminate the scale effect. However, the implementation of LOAM is not normalized by the distance. 
https://www.ri.cmu.edu/wp-content/uploads/2019/01/AURO_2017_2.pdf
[image]
Any idea the reason for the implementation?

WaldenLeefx, 2021-11-10:
@DarrenWong Do you know where this equation comes from? Does it come from discrete Laplace operator? https://en.wikipedia.org/wiki/Discrete_Laplace_operator

DarrenWong, 2021-11-10:
Do you mean the ||.|| operator? It should be L2 norm  https://en.wikipedia.org/wiki/Norm_(mathematics)

closed 2021-11-10 by DarrenWong

WaldenLeefx, 2021-11-10:
Thanks a lot. I mean the equation (1）mentioned in your issues. I want to write a report about how to calculate curvature in different ways. It seems that equation (1) has nothing to do with differential geometry, right?

DarrenWong, 2021-11-12:
@WaldenLeefx yes, it uses a general form to detect edges and planar points. What do you mean by differential geometry? different geometry?